AravisProject/aravis #874: arv_camera_gv_auto_packet_size() crash bug

CynthiaLiLi, 2024-02-27:
**Describe the bug**
Calling arv_camera_gv_auto_packet_size() to automatically negotiate the packet size causes the program to crash.

**Camera description:**
 - Manufacturer[Hikrobot]
 - Model[MV-CB060-10GM-S-W]
 - Interface [GIGE]

**Platform description:**
 - Aravis version [0.8.29]
 - OS [windows]
 - Hardware [x86_64]
 - Development Framework [QT]

**Additional context**
stack information:
[image]

EmmanuelP, 2024-02-27:
Thanks for the report.

What does mean the last but one column ?

Is the crash reproducible ? If so, could you try to capture a complete backtrace of the crash, with the variable values, similar to what can be obtained using gdb ?

CynthiaLiLi, 2024-02-27:
这是来自QQ邮箱的假期自动回复邮件。
 
你好，我最近正在休假中，无法亲自回复你的邮件。我将在假期结束后，尽快给你回复。

CynthiaLiLi, 2024-02-29:
1. What does mean the last but one column ?
> The last line is a function I wrote myself to initialize the camera. In this function, arv_camera_gv_auto_packet_size() is called.
> In addition, I did another test.
After replacing arv_camera_gv_auto_packet_size() with arv_camera_gv_set_packet_size(), if I set a value that exceeds the maximum supported by the camera, the program still crashes when calling arv_camera_create_stream(). The following is the stack information:
[image]

2. Is the crash reproducible ? If so, could you try to capture a complete backtrace of the crash, with the variable values, similar to what can be obtained using gdb ?
> Yes, the crash can be reproduced.
 I am currently using a compiled dll file in the program, so I cannot trace back more information. When I have time later, I can download the aravis source code for further debugging and locating problems.

EmmanuelP, 2024-02-29:
> >  What does mean the last but one column ?
> The last line is a function I wrote myself to initialize the camera. In this function, arv_camera_gv_auto_packet_size() is called.

That is not what I'm asking. I was wondering the meaning of the numbers in the last but one column (770, 796, ...).