AravisProject/aravis #882: Documentation request: documentation for each function should state how/whether to free the output

dkogan, 2024-03-12:
Hi. I'm starting out, and it's not clear what should or should not be freed. In particular I was doing this:

```
    ArvGcNode* feature = get_string_or_enum_feature(key, device);
    ....
    g_object_unref(feature);
```

I wasn't getting any crashes, but things wouldn't work in mysterious ways. Apparently freeing the feature wasn't needed, and the unref was breaking stuff. It would be really good if the docs for each function were clear on this point.

Thank you very much for aravis!

EmmanuelP, 2024-03-13:
Hi.

Do you have a precise example ?

`arv_device_get_feature()` seems correctly documented: [https://aravisproject.github.io/aravis/aravis-stable/method.Device.get_feature.html](https://aravisproject.github.io/aravis/aravis-stable/method.Device.get_feature.html)

dkogan, 2024-03-14:
Hi. I guess there're multiple kinds of similar-but-not identical documentation. I was looking at this:

https://aravisproject.github.io/docs/aravis-0.8/ArvDevice.html#arv-device-get-feature

It's missing the info about ownership. Those docs were linked from the main page. Should this be consolidated?

Thanks

EmmanuelP, 2024-03-14:
I have removed the link to this obsolete documentation from the main page. I don't even know how it comes there is a content at this location.

EmmanuelP, 2024-03-14:
By the way, the old documentation has the information (transfer none). This was made clearer with gi-docgen.

closed 2024-03-14 by EmmanuelP

dkogan, 2024-03-14:
Awesome; thanks. I just stumbled on two more documentation quirks. Fixing these should be easy, and would make it easier for new users:

- The top-level page on the documentation server: https://aravisproject.github.io/ doesn't have any useful information or any links to anything that isn't itself. Ideally it should redirect to the docs, or the docs should live there directly
- I'm looking at the docs for arv_device_write_memory(): https://aravisproject.github.io/aravis/aravis-stable/method.Device.write_memory.html. It says about the "buffer" argument: "The instance takes ownership of the data, and is responsible for freeing it". This isn't right, I'm guessing? If it was, then you wouldn't be able to pass in data on the stack, for instance.
Thanks

dkogan, 2024-03-14:
Oh, and if that snippet is indeed wrong, can the buffer be made const?