AravisProject/aravis #892: ArvPixelFormat wrong values

1vanKa, 2024-03-29:
It looks like there is a mistake in `ARV_PIXEL_FORMAT_BAYER_RG_8` and `ARV_PIXEL_FORMAT_BAYER_BG_8`.

arv_pixel_format_to_gst_caps_string works correctly saying that 0x01080009u is bayerRGGB (which is also called bayerBG), but `ARV_PIXEL_FORMAT_BAYER_BG_8` value is 0x0108000bu. Also OpenCV correctly interprets images from my camera as bayerBG images, but bayerRG looks weird (like red and blue channels swapped).

I don’t know is 0x0108000bu correct for bayerRG, but I can say that 0x01080009u is bayerBG.

Camera:
TIS DFK 33gx290
Interface: GigE

These values are in main branch of aravis.

jeffwitz, 2024-03-29:
OpenCV use BGR not RGB.
img=img[::,::,::-1] should do the trick

1vanKa, 2024-04-01:
1. I'm using C api;
2. I know, that opencv uses BGR, but it also handles it internally when, for example, using `cv::imwrite` or `cv::imencode`;
3. If I convert image with `cv::cvtColor(img, convertedImage, cv::COLOR_BayerBG2BGR);` image looks fine, but `arv_camera_get_pixel_format` returns `0x01080009u` which is `ARV_PIXEL_FORMAT_BAYER_RG_8`.

1vanKa, 2024-04-02:
Sorry for everyone, I was wrong. My camera uses BayerRG format. There is something weird in openCV COLOR_Bayer* naming.

closed 2024-04-02 by 1vanKa