AravisProject/aravis #893: Bug with payload

jeffwitz, 2024-03-29:
**Describe the bug**
wrong buffer size for payload definition

**To Reproduce**
```python
import gi
gi.require_version('Aravis', '0.8')
from gi.repository import Aravis

Aravis.update_device_list()

print(Aravis.get_device_id(0))
cam = Aravis.Camera.new(Aravis.get_device_id(0))

# Vérifiez et affichez la configuration actuelle de la caméra
print("Format de pixel:", cam.get_pixel_format_as_string())
print("Résolution:", cam.get_width_bounds()[1], "x", cam.get_height_bounds()[1])

payload_size = cam.get_payload()
print("payload size:", payload_size, "octets")

print("(payload size)/(width*height):")
print(payload_size/(cam.get_width_bounds()[1]*cam.get_height_bounds()[1]))

stream = cam.create_stream(None, None)

# Allocation et ajout d'un buffer basé sur la taille correcte du payload
buffer = Aravis.Buffer.new_allocate(payload_size)
stream.push_buffer(buffer)

cam.start_acquisition()

try:
    buf = stream.timeout_pop_buffer(5000000)
    if buf.get_status() == Aravis.BufferStatus.SUCCESS:
        print("Buffer reçu avec succès.")
    else:
        print("Buffer non reçu ou erreur de buffer:", buf.get_status())
finally:
    cam.stop_acquisition()
```

gives 

`<enum ARV_BUFFER_STATUS_SIZE_MISMATCH of type Aravis.BufferStatus>`

**Expected behavior**
size match

**Camera description:**
 - The Imaging Source Europe GmbH-199E48910488-48910488
 - Interface [e.g. USB3]

**Platform description:**
 - Aravis version 
 - OS: debian 12
 - Hardware [e.g. x86_64]

**Additional context**
works with arv-view

EmmanuelP, 2024-06-16:
Please attach the console output when running your test software with `ARV_DEBUG` set to `all`.

EmmanuelP, 2024-08-14:
I have just tested your test code with the fake camera and a PointGrey camera. It worked in both cases.

I'm closing this issue for now. Don't hesitate to reopen once you have th request informations.

Thanks.

closed 2024-08-14 by EmmanuelP