AravisProject/aravis #899: not able to read chunk data using aravis 0.8 (it worked with 0.6)

weichselbaumj, 2024-05-03:
**Describe the bug**
Since upgrade to aravis 0.8 (from 0.6) reading chunk data does not work. I also tried the given example code (https://github.com/AravisProject/aravis/blob/main/tests/arvchunkparsertest.c), it does not work in my configuration.

**To Reproduce**
- compile https://github.com/AravisProject/aravis/blob/main/tests/arvchunkparsertest.c
- connect camera (FLIR-Blackfly S BFS-PGE-16S2C)
- start the test program

**Expected behavior**
Getting desired chunk data.

**Camera description:**
 - FLIR
 - FLIR-Blackfly S BFS-PGE-16S2C
 - GigE

**Platform description:**
 - Aravis version 0.8.30
 - OS: Ubuntu 20.04
 - x86_64

**Additional context**
Reading chunk data worked fine with aravis 0.6. Fore some reason we had to upgrade to aravis 0.8 and besides of the chunk data issue, everything works as expected.
Thanks for your help.

weichselbaumj, 2024-05-03:
Additional output by using ARV_DEBUG=all

```
[12:58:50.396] 🅸 interface> Found 0 USB3Vision device (among 9 USB devices)
[12:58:50.397] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[12:58:50.397] 🅸 interface> [GvDiscoverSocket::new] Add interface 10.101.41.6 (10.101.41.255)
[12:58:50.398] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.130.1 (192.168.130.255)
[12:58:50.398] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.86.17 (192.168.86.255)
[12:58:50.398] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.120.1 (192.168.120.255)
[12:58:50.398] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.110.1 (192.168.110.255)
[12:58:50.398] 🅸 interface> [GvDiscoverSocket::new] Add interface 172.17.0.1 (172.17.255.255)
[12:58:50.398] 🅸 interface> [GvDiscoverSocket::new] Add interface 10.9.0.6 (10.9.0.255)
[12:58:50.400] 🅸 interface> [GvInterface::discovery] Device 'FLIR-Blackfly S BFS-PGE-16S2C-22458335' found (interface 192.168.110.1) user_id '' - MAC '2c:dd:a3:56:af:df'
[12:58:50.400] 🅸 device> [GvDevice::new] Interface address = 192.168.110.1
[12:58:50.400] 🅸 device> [GvDevice::new] Device address = 192.168.110.2
[12:58:50.407] 🅸 device> [GvDevice::load_genicam] xml url = 'Local:BFS-PGE-16S2C_019ADA4C.zip;60000000;13DBC' at 0x200
[12:58:50.407] 🅸 device> [GvDevice::load_genicam] Xml address = 0x60000000 - size = 0x13dbc - BFS-PGE-16S2C_019ADA4C.zip
[12:58:50.600] 🅸 device> [GvDevice::load_genicam] Zipped xml data
[12:58:50.606] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'xmlns:xi'
[12:58:50.694] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationLLA'
[12:58:50.694] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationDHCP'
[12:58:50.694] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationPersistentIP'
[12:58:50.694] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPAddress'
[12:58:50.694] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentSubnetMask'
[12:58:50.694] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentDefaultGateway'
[12:58:50.694] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevNumberOfNetworkInterfaces'
[12:58:50.694] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentIPAddress'
[12:58:50.694] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentSubnetMask'
[12:58:50.694] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentDefaultGateway'
[12:58:50.694] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelCount'
[12:58:50.694] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelSelector'
[12:58:50.694] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPAddrCalc'
[12:58:50.694] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevTimestampTickFrequency'
[12:58:50.695] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPHostPort'
[12:58:50.695] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSFireTestPacket'
[12:58:50.695] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSDoNotFragment'
[12:58:50.695] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSBigEndian'
[12:58:50.695] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSPacketSize'
[12:58:50.695] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPD'
[12:58:50.695] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCDA'
[12:58:50.695] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCSP'
[12:58:50.695] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCCFGMultipart'
[12:58:50.698] 🅸 device> [GvDevice::new] Device endianness = big
[12:58:50.698] 🅸 device> [GvDevice::new] Packet resend     = yes
[12:58:50.698] 🅸 device> [GvDevice::new] Write memory      = yes
[12:58:50.698] 🅸 device> [GvDevice::new] Legacy endianness handling = yes
[12:58:50.700] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'xmlns:xi'
[12:58:51.469] 🅸 device> [GvDevice::create_stream] Number of stream channels = 1
[12:58:51.486] 🅸 device> [GvDevice::auto_packet_size] Current packet size check successfull (5000 bytes)
[12:58:51.491] 🅸 stream> [GvStream::stream_new] Stream channel = 0
[12:58:51.494] 🅸 stream> [GvStream::stream_new] Packet size = 5000 byte(s)
[12:58:51.498] 🅸 stream> [GvStream::stream_new] Destination stream port = 53823
[12:58:51.498] 🅸 stream> [GvStream::stream_new] Source stream port = 49153
[12:58:51.498] 🅸 stream> [GvStream::loop] Standard socket method

** (process:561348): CRITICAL **: 12:58:51.538: arv_buffer_get_chunk_data: assertion 'arv_buffer_has_chunks (buffer)' failed
[12:58:51.538] 🆆 chunk> [ChunkWidth] [ChunkWidth_Val] [WidthPort] Chunk 0x05040006 not found
ChunkWidth = 0
error getting chunk data:
[ChunkWidth] [ChunkWidth_Val] [WidthPort] Chunk 0x05040006 not found
[12:58:51.545] 🅸 stream>     bins    ;frame_retent; packet_time;inter_packet
                                    0;           0;          44;         471
                                 2000;           0;          50;           0
                                 4000;           0;          50;           0
                                 6000;           0;          50;           0
                                 8000;           0;          50;           0
                                10000;           0;          50;           0
                                12000;           0;          50;           0
                                14000;           0;          50;           0
[... cut ...]
[12:58:51.546] 🅸 stream> [GvStream::finalize] n_resend_ratio_reached = 0
[12:58:51.546] 🅸 stream> [GvStream::finalize] n_resend_disabled      = 0
[12:58:51.546] 🅸 stream> [GvStream::finalize] n_duplicated_packets   = 0
[12:58:51.546] 🅸 stream> [GvStream::finalize] n_transferred_bytes    = 2336660
[12:58:51.546] 🅸 stream> [GvStream::finalize] n_ignored_bytes        = 517132
[12:58:51.546] 🅸 stream> [Stream::finalize] Flush 0 buffer[s] in input queue
[12:58:51.546] 🅸 stream> [Stream::finalize] Flush 0 buffer[s] in output queue
```

EmmanuelP, 2024-05-03:
Could you attach the genicam data of your device ?

weichselbaumj, 2024-05-06:
Here are the outputs of `arv-tool-0.8 values` and `arv-tool-0.8 features`. If that is not what you meant, please let me know. Thanks.
[features.txt](https://github.com/AravisProject/aravis/files/15217656/features.txt)
[values.txt](https://github.com/AravisProject/aravis/files/15217657/values.txt)

EmmanuelP, 2024-06-16:
I can not reproduce the issue here using a BFLY-PGE-14S2C-CS.

```
Sources/aravis/build on  aravis-0-8 [$] took 4s ❯ ARV_DEBUG=all ./tests/arv-chunk-parser-test 
[09:18:17.492] 🅸 interface> Found 0 USB3Vision device (among 16 USB devices)
[09:18:17.492] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[09:18:17.492] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.0.114 (192.168.0.255)
[09:18:17.494] 🅸 interface> [GvInterface::discovery] Device 'Point Grey Research-Blackfly BFLY-PGE-14S2C-13125101' found (interface 192.168.0.114) user_id 'PointGrey' - MAC '00:b0:9d:c8:45:ed'
[09:18:17.494] 🅸 device> [GvDevice::new] Interface address = 192.168.0.114
[09:18:17.494] 🅸 device> [GvDevice::new] Device address = 192.168.0.93
[09:18:17.496] 🅸 device> [GvDevice::load_genicam] xml url = 'Local:GRS_GEV_v003_217679.zip;7F1D0040;8474' at 0x200
[09:18:17.496] 🅸 device> [GvDevice::load_genicam] Xml address = 0x7f1d0040 - size = 0x8474 - GRS_GEV_v003_217679.zip
[09:18:17.606] 🅸 device> [GvDevice::load_genicam] Zipped xml data
[09:18:17.608] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'xmlns:xi'
[09:18:17.644] 🅸 genicam> [Gc::set_default_node_data] Add 'DeviceManufacturerInfo'
[09:18:17.644] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationLLA'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationDHCP'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationPersistentIP'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPAddress'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentSubnetMask'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentDefaultGateway'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevNumberOfNetworkInterfaces'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentIPAddress'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentSubnetMask'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentDefaultGateway'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelCount'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelSelector'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPAddrCalc'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevTimestampTickFrequency'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPHostPort'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSFireTestPacket'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSDoNotFragment'
[09:18:17.645] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSBigEndian'
[09:18:17.646] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSPacketSize'
[09:18:17.646] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPD'
[09:18:17.646] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCDA'
[09:18:17.646] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCSP'
[09:18:17.646] 🅸 genicam> [Gc::set_default_node_data] Add 'ArvGevSCCFGMultipart'
[09:18:17.649] 🅸 device> [GvDevice::new] Device endianness = little
[09:18:17.649] 🅸 device> [GvDevice::new] Packet resend     = yes
[09:18:17.650] 🅸 device> [GvDevice::new] Write memory      = yes
[09:18:17.650] 🅸 device> [GvDevice::new] Legacy endianness handling = yes
[09:18:17.652] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'xmlns:xi'
[09:18:17.933] 🅸 device> [GvDevice::create_stream] Number of stream channels = 1
[09:18:17.951] 🅸 device> [GvDevice::auto_packet_size] Current packet size check successfull (1400 bytes)
[09:18:17.955] 🅸 stream> [GvStream::stream_new] Stream channel = 0
[09:18:17.959] 🅸 stream> [GvStream::stream_new] Packet size = 1400 byte(s)
[09:18:17.964] 🅸 stream> [GvStream::stream_new] Destination stream port = 58679
[09:18:17.964] 🅸 stream> [GvStream::stream_new] Source stream port = 1030
[09:18:17.964] 🅸 stream> [GvStream::loop] Standard socket method
ChunkWidth = 1296
ChunkHeight = 1032
[09:18:18.004] 🅸 stream>     bins    ;frame_retent; packet_time;inter_packet
                                    0;           0;         134;         982
                                 2000;           0;         135;           0
                                 4000;           0;         140;           0
                                 6000;           0;         133;           0
                                 8000;           0;         141;           0
                                10000;           0;         141;           0
                                12000;           0;         132;           0
                                14000;           1;          27;           0
                         -------------
                         >=     16000;           0;           0;           0
                         <          0;           0;           0;           0
                         min         :       14470;           0;           0
                         max         :       14470;       14470;         368
                         last max at :           0;         982;         695
                         counter     :           1:         983:         982
[09:18:18.004] 🅸 stream> [GvStream::finalize] n_completed_buffers    = 1
[09:18:18.004] 🅸 stream> [GvStream::finalize] n_failures             = 0
[... cut ...]
[09:18:18.004] 🅸 stream> [GvStream::finalize] n_resend_ratio_reached = 0
[09:18:18.004] 🅸 stream> [GvStream::finalize] n_resend_disabled      = 0
[09:18:18.004] 🅸 stream> [GvStream::finalize] n_duplicated_packets   = 0
[09:18:18.004] 🅸 stream> [GvStream::finalize] n_transferred_bytes    = 1345390
[09:18:18.004] 🅸 stream> [GvStream::finalize] n_ignored_bytes        = 223656
[09:18:18.004] 🅸 stream> [Stream::finalize] Flush 0 buffer[s] in input queue
[09:18:18.004] 🅸 stream> [Stream::finalize] Flush 0 buffer[s] in output queue
```

EmmanuelP, 2024-06-16:
```
Category    : 'Root'
    Category    : 'AnalogControl'
        Enumeration  : [RO] 'GainSelector'
              * Gain
              * GainAuto
              * AutoGainLowerLimit
              * AutoGainUpperLimit
            EnumEntry   : 'All'
        Enumeration  : [RW] 'GainAuto'
            EnumEntry   : 'Continuous'
            EnumEntry   : 'Once'
            EnumEntry   : 'Off'
        Float        : [RW] 'Gain'
        Float        : [RW] 'AutoGainLowerLimit'
        Float        : [RW] 'AutoGainUpperLimit'
        Float        : [RW] 'BlackLevel'
        Boolean      : [RW] 'BlackLevelEnabled'
        Float        : [RW] 'Gamma'
        Boolean     : 'GammaEnabled' (Not available)
        Integer      : [RW] 'Sharpness'
        Boolean     : 'SharpnessEnabled' (Not available)
        Enumeration : 'SharpnessAuto' (Not available)
            EnumEntry   : 'Continuous' (Not available)
            EnumEntry   : 'Once' (Not available)
            EnumEntry   : 'Off' (Not available)
        Float        : [RW] 'Hue'
        Boolean     : 'HueEnabled' (Not available)
        Float        : [RW] 'Saturation'
        Boolean     : 'SaturationEnabled' (Not available)
        Enumeration : 'SaturationAuto' (Not available)
            EnumEntry   : 'Off'
        Enumeration  : [RW] 'BalanceWhiteAuto'
            EnumEntry   : 'Continuous'
            EnumEntry   : 'Once'
            EnumEntry   : 'Off'
        Enumeration  : [RW] 'BalanceRatioSelector'
              * BalanceRatio
            EnumEntry   : 'Blue'
            EnumEntry   : 'Red'
        Float       : 'BalanceRatio' (Not available)
    Category    : 'DeviceControl'
        StringReg    : [RO] 'DeviceVendorName'
        StringReg    : [RO] 'DeviceModelName'
        StringReg    : [RO] 'DeviceVersion'
        StringReg    : [RO] 'DeviceSerialNumber'
        StringReg    : [RO] 'DeviceID'
        StringReg    : [RW] 'DeviceUserID'
        StringReg    : [RO] 'pgrSensorDescription'
        StringReg    : [RO] 'DeviceSVNVersion'
        StringReg    : [RO] 'DeviceFirmwareVersion'
        Enumeration  : [RO] 'DeviceScanType'
            EnumEntry   : 'Areascan'
        Float        : [RO] 'DeviceTemperature'
        Command      : [WO] 'DeviceReset'
        Integer      : [RO] 'pgrDeviceUptime'
        Enumeration  : [RW] 'AutoFunctionAOIsControl'
              * AutoFunctionAOIWidth
              * AutoFunctionAOIHeight
              * AutoFunctionAOIOffsetX
              * AutoFunctionAOIOffsetY
            EnumEntry   : 'On'
            EnumEntry   : 'Off'
        Integer     : 'AutoFunctionAOIWidth' (Not available)
        Integer     : 'AutoFunctionAOIHeight' (Not available)
        Integer     : 'AutoFunctionAOIOffsetX' (Not available)
        Integer     : 'AutoFunctionAOIOffsetY' (Not available)
        Enumeration  : [RW] 'pgrDevicePowerSupplySelector'
              * pgrPowerSourcePresent
              * pgrPowerSupplyEnable
[... 382 lines cut ...]
        Enumeration  : [RW] 'EventSelector'
            EnumEntry   : 'ExposureEnd'
            EnumEntry   : 'AcquisitionEnd'
            EnumEntry   : 'AcquisitionStart'
        Enumeration  : [RW] 'EventNotification'
            EnumEntry   : 'Off'
            EnumEntry   : 'On'
        Category    : 'EventAcquisitionStartData'
            Integer      : [RO] 'EventAcquisitionStart'
            Integer      : [RO] 'EventAcquisitionStartTimestamp'
            Integer      : [RO] 'EventAcquisitionStartFrameID'
        Category    : 'EventAcquisitionEndData'
            Integer      : [RO] 'EventAcquisitionEnd'
            Integer      : [RO] 'EventAcquisitionEndTimestamp'
            Integer      : [RO] 'EventAcquisitionEndFrameID'
        Category    : 'EventExposureEndData'
            Integer      : [RO] 'EventExposureEnd'
            Integer      : [RO] 'EventExposureEndTimestamp'
            Integer      : [RO] 'EventExposureEndFrameID'
    Category    : 'RemoveParameterLimits'
        Enumeration  : [RW] 'ParameterSelector'
              * RemoveLimits
            EnumEntry   : 'Gain'
        Boolean      : [RW] 'RemoveLimits'
```

EmmanuelP, 2024-06-16:
I don't see any difference in the way chunk are defined between the 2 devices.

EmmanuelP, 2024-06-16:
Please attach the defintions of `ChunkWidth` and `ChunkHeight` in the genicam data of your camera.

weichselbaumj, 2024-06-18:
I can confirm that reading Chunk data with a `BFLY-PGE-14S2C-CS` works as expected.

The output of `arv-tool-0.8 features` for both cameras can be found here:
[BFLY-PGE-14S2C.txt](https://github.com/user-attachments/files/15885253/BFLY-PGE-14S2C.txt)
[BFS-PGE-16S2C.txt](https://github.com/user-attachments/files/15885256/BFS-PGE-16S2C.txt)

I hope this helps. Please let me know if you need more information.
Thanks a lot.

EmmanuelP, 2024-08-14:
I still don't understand :(

Could you attach the genicam data of the BFS-PGE-16S2C camera ?

weichselbaumj, 2024-08-26:
Hi @EmmanuelP 
I already attached the Genicam data in this post. If this is not what you need, could you please give me instructions how to get the data you need?
Thanks a lot in advance.
BR Johann

> Here are the outputs of `arv-tool-0.8 values` and `arv-tool-0.8 features`. If that is not what you meant, please let me know. Thanks. [features.txt](https://github.com/AravisProject/aravis/files/15217656/features.txt) [values.txt](https://github.com/AravisProject/aravis/files/15217657/values.txt)