AravisProject/aravis #903: aravis 0.10 not stream

flightiger, 2024-05-17:
I used the LEO25000S-24GC camera from Vision Datum company and deployed Aravis via MSYS2. I was able to successfully detect and connect to the device. However, the window does not refresh with images. The camera can be pinged, but I'm unsure where the problem lies. Can you give me some guidance?

# arv-camera-test-0.10 -d all
Looking for the first available camera
[14:49:38.447] ≡ƒà╕ interface> Found 0 USB3Vision device (among 6 USB devices)
[14:49:38.447] ≡ƒà╕ interface> [GvDiscoverSocket::new] Add interface 192.168.170.1 (192.168.170.255)
[14:49:38.447] ≡ƒà╕ interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.255.255.255)
[14:49:38.447] ≡ƒà╕ interface> [GvInterface::discovery] Device 'Vision Datum-LEO2 5000S-24gc-DA09814
00' found (interface 192.168.170.1) user_id '' - MAC '34:bd:20:2a:62:b3'
[14:49:38.447] ≡ƒà╕ device> [GvDevice::new] Interface address = 192.168.170.1
[14:49:38.447] ≡ƒà╕ device> [GvDevice::new] Device address = 192.168.170.2
[14:49:38.447] ≡ƒà╕ device> [GvDevice::load_genicam] xml url = 'Local:Vision Datum_GigE_Vision_1_V3_
4_54.zip;100000;1699c' at 0x200
[14:49:38.463] ≡ƒà╕ device> [GvDevice::load_genicam] Xml address = 0x100000 - size = 0x1699c - Visio
n Datum_GigE_Vision_1_V3_4_54.zip
[14:49:38.633] ≡ƒà╕ device> [GvDevice::load_genicam] Zipped xml data
[14:49:38.633] ≡ƒà╕ dom> [GcFeature::set_attribute] Unknown attribute 'xmlns:mstns'
[14:49:38.665] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationLLA'
[14:49:38.665] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationDHCP'
[14:49:38.665] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationPersistent
IP'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPAddress'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentSubnetMask'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentDefaultGateway'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevNumberOfNetworkInterfaces'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentIPAddress'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentSubnetMask'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentDefaultGateway'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelCount'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelSelector'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevSCPAddrCalc'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevTimestampTickFrequency'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevSCPHostPort'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSFireTestPacket'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSDoNotFragment'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSBigEndian'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSPacketSize'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevSCPD'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevSCDA'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevSCSP'
[14:49:38.680] ≡ƒà╕ genicam> [Gc::set_default_node_data] Add 'ArvGevSCCFGMultipart'
[14:49:38.680] ≡ƒà╕ device> [GvDevice::new] Device endianness = big
[14:49:38.680] ≡ƒà╕ device> [GvDevice::new] Packet resend     = yes
[14:49:38.680] ≡ƒà╕ device> [GvDevice::new] Write memory      = yes
[14:49:38.680] ≡ƒà╕ device> [GvDevice::new] Legacy endianness handling = yes
vendor name            = Vision Datum
model name             = LEO2 5000S-24gc
device serial number   = DA0981400
image width            = 1600
image height           = 1000
horizontal binning     = 1
vertical binning       = 1
exposure               = 120 ┬╡s
gain                   = 3 dB
payload                = 3200000 bytes
gv n_stream channels   = 1
gv current channel     = 0
gv packet delay        = 4000 ns
[14:49:38.758] ≡ƒà╕ device> [GvDevice::create_stream] Number of stream channels = 1
[14:49:38.774] ≡ƒà╕ device> [GvDevice::auto_packet_size] Current packet size check successfull (8164
 bytes)
[14:49:38.774] ≡ƒà╕ stream> [GvStream::stream_new] Stream channel = 0
[14:49:38.774] ≡ƒà╕ stream> [GvStream::stream_new] Packet size = 8164 byte(s)
[14:49:38.790] ≡ƒà╕ stream> [GvStream::stream_new] Destination stream port = 58710
[14:49:38.790] ≡ƒà╕ stream> [GvStream::stream_new] Source stream port = 9001
gv packet size         = 8164 bytes
[14:49:38.790] ≡ƒà╕ stream> [GvStream::loop] Standard socket method
[14:49:38.805] ≡ƒåå stream-thread> [GvStream::loop] receive_messages failed: A non-blocking socket o
peration could not be completed immediately.
[14:49:38.805] ≡ƒåå stream-thread> [GvStream::loop] receive_messages failed: A non-blocking socket o
[... 53 lines cut ...]
                         <          0;           0;           0;           0
                         min         :       20347;           0;           0
                         max         :       42871;       42871;        4384
                         last max at :           4;        1979;        1974
                         counter     :          24:        9290:        9266
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_completed_buffers    = 23
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_failures             = 0
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_underruns            = 0
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_timeouts             = 0
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_aborted              = 1
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_missing_frames       = 0
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_size_mismatch_errors = 0
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_received_packets     = 9290
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_missing_packets      = 0
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_error_packets        = 0
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_ignored_packets      = 0
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_resend_requests      = 15
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_resent_packets       = 15
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_resend_ratio_reached = 0
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_resend_disabled      = 0
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_duplicated_packets   = 0
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_transferred_bytes    = 75146582
[14:51:03.726] ≡ƒà╕ stream> [GvStream::finalize] n_ignored_bytes        = 0
[14:51:03.726] ≡ƒà╕ stream> [Stream::delete_buffers] Delete 50 buffer[s] in input queue
[14:51:03.726] ≡ƒà╕ stream> [Stream::delete_buffers] Delete 0 buffer[s] in output queue

EmmanuelP, 2024-06-16:
Hi,

The output of `arv-camera-test` shows a perfectly working acquisition.

Are you trying to use `arv-viewer` on Windows ? What does the console output say ? May be there is missing gstreamer packages ?

flightiger, 2024-06-16:
Hithank you very much for your reply. After several attempts, I successfully deployed Aravis on the Ubuntu 22.04 system, and it is working well. However, on the Windows system, it seems that there are always some missing dependencies. I will review my experiment again, and if I have any new findings, I will reply to you at the first opportunity. Wishing you all the best.kevin