AravisProject/aravis #904: aravis-viewer or library is giving error for the image region size above 640x480 with cypress FX3 camera

HarshilSoni849, 2024-05-22:
**Describe the bug**
We are using cypress FX3 USB3 camera for our project purpose , when we use aravis-viewer or any other tool to see streaming then it is giving weird scenario.
With image region size 640x480 it is working properly
but with the size above then 640x480 like 1920x1080 or 720x480 some fix number of images are getting captured like 4 or 9 then without giving any error streaming stops.

**To Reproduce**
connect cypress FX3 camera ,
open arv-viewer binary 
select region size 1920x1080 and observe 
then select 640x480 and observe

**Expected behavior**
it should work with size higher then 640x480

**Camera description:**
-  cypress FX3 USB3
- manufacturer : infineon

**Platform description:**
 - Aravis version - 0.10 , 0.8
 - OS: ubuntu 20.04
 - PC

HarshilSoni849, 2024-05-31:
hello there ,

is there any update for me ?
for the query I generated ?

EmmanuelP, 2024-06-16:
Hi,

Try to run `arv-camera-test` with `-d all` and the different resolutions, and report the console output here.

HarshilSoni849, 2024-06-17:
Hello ,

I can not able to find any console output ,
have you attached any file ?

EmmanuelP, 2024-08-14:
No. You should open a terminal, and launch `arv-camera-test-0.8 -d all` from there. Then copy paste here what is printed in the terminal window.

I'm closing the issue for now. Don't hesitate to reopen once you have the requested information.

closed 2024-08-14 by EmmanuelP