AravisProject/aravis #906: ValueError: Pointer arguments are restricted to integers, capsules, and None. See: https://bugzilla.gnome.org/show_bug.cgi?id=683599

hk-du, 2024-06-10:
**Describe the bug**
Creating a preallocated buffer in python throws the above error 

**To Reproduce**
import gi
gi.require_version('Aravis','0.8')
from gi.repository import Aravis, 

fake_array = numpy.arange(1,17, dtype= numpy.uint16).reshape((4,4))
fake_buffer = Aravis.Buffer.new(fake_array.nbytes, fake_array.flatten().tobytes())

**Expected behavior**
I expected to have an Aravis.Buffer reference to be created 

**Camera description:**
 - arv-fake-gv-camera-0.8 -i 127.0.0.1
 

**Platform description:**
 - Aravis version 0.8
 - OS: [e.g. Ubuntu 22.04]
 - Hardware [x86_64]
 - Python version: Python 3.10.12 64bit

**Additional context**
Trying in python to create an Aravis Buffer which is already preallocated and populated with fake values like above. The context is to push this buffer into a stream, start acquiring with a fake camera, and see if the values can be correctly read out once the buffer is poped.

EmmanuelP, 2024-06-16:
Thanks for the report.

Could you please attach a simple python script that shows the issue ?

EmmanuelP, 2024-08-14:
No update, I'm closing the issue. Don't hesitate to reopen with the requested information.

Thanks.

closed 2024-08-14 by EmmanuelP

jimustafa, 2026-02-21:
Greetings @EmmanuelP! I am running into the same error when trying to use the `read_memory` method on `Aravis.FakeDevice` object.

Suppose I have
```python
import gi

gi.require_version("Aravis", "0.8")
from gi.repository import Aravis

Aravis.enable_interface("Fake")

camera = Aravis.Camera.new("Fake_1")
device = camera.get_device()
```

and I want to do
```python
# read pixel format
pixel_format = device.read_memory(0x128, 4, buffer=???)
```
What needs to be passed as the `buffer` argument?

Is this even possible? Probably just missing something simple... Thanks for the help!