AravisProject/aravis #911: chunk parser, setting of chunk selector nodes

SebastianRiedel, 2024-07-11:
### Description

Follow-up to the [How to set a Chunk Selector node?](https://aravis-project.discourse.group/t/how-to-set-a-chunk-selector-node/787) discord discussion.

It is a about accessing scale and offset values per channel ABC for conversion of point cloud data in the abstract format `Coord3D_ABC16`  to metric values like [mm].

Noteworthy documentation from Mechmind:
- [HALCON Genicam examples](https://github.com/MechMindRobotics/mecheye_halcon_samples/blob/main/area_scan_3d_camera/obtain_point_cloud_with_normals.hdev)
- Mechmind Documentation on [Obtaining Point Cloud Data from Depth Map in Custom Development with GenICam](https://community.mech-mind.com/t/topic/2061)

What I can confirm at this point using the code snippet in the [How to set a Chunk Selector node?](https://aravis-project.discourse.group/t/how-to-set-a-chunk-selector-node/787) original post is:
- I get the exact same offset for all coordinates ABC
- offset and scale change per image acquisition, so they are not static

**Describe the solution you'd like**

**Describe alternatives you've considered**
- [ ] todo: provide minimal sample code and output to further elicit the problem
- [ ] todo: double check to confirm proposed approach in discord doesn't work, cf. https://aravis-project.discourse.group/t/how-to-set-a-chunk-selector-node/787/2?u=sebastianriedel

### Additional context
arv-tool 0.8.30: genicam xml and features.txt attached: 
[arv_tool_mechmind_nano.zip](https://github.com/user-attachments/files/16178992/arv_tool_mechmind_nano.zip)

EmmanuelP, 2024-07-12:
Hi,

Please try #912 

The Genicam data look wrong though. The access to the offset chunk only uses ChunkComponentSelector. A ChunkScan3dCoordinateSelector feature is defined but never used for chunk data access. I don't see how what is described in your camera documentation could work.

SebastianRiedel, 2024-08-20:
I'm sorry, priorities shifting a lot and everything is urgent... I want to test it again in order to give an update, but need to find the time still.