AravisProject/aravis #921: Aravis viewer endianness issue with mono 16-bit pixel format

berke, 2024-08-24:
Images are incorrectly displayed in Mono16 mode due to the endianness being wrong.

**To Reproduce**
- Run Aravis viewer
- Select Mono16 pixel format

**Expected behavior**
Image displays properly.

**Camera description:**
 - Manufacturer: Point Grey Research
 - Model: Blackfly BFLY-PGE-31S4M
 - Interface: GigE

**Platform description:**
 - Aravis version: 0.8.26-1 (Debian package), 0.10 (commit `2ebaa8661761...`)
 - OS: Debian 12
 - Hardware: x86_64

**Additional context**
- Changing `pgrPixelBigEndian` has no effect
- Changing the GStreamer format description from `GRAY16_LE` to `GRAY16_BE` in `arvmisc.c` solves the issue for me, but a proper solution may or may not require a kludge or an option in the UI to force endianness.
```
	{
		ARV_PIXEL_FORMAT_MONO_16,
		"video/x-raw, format=(string)GRAY16_BE",
		"video/x-raw",		"GRAY16_BE",
		"video/x-raw-gray, bpp=(int)16, depth=(int)16",
		"video/x-raw-gray",	16,	16,	0
	},
```

EmmanuelP, 2024-08-25:
Hi,

In the blackfly documentation, there is a `PixelBigEndian` feature. Isn't this feature set to False by default ?

EmmanuelP, 2024-08-25:
For the record, I have the same issue with a BFLY-PGE-14S2C-CS. It looks like an issue with the firmware, it seems they should send little endian pixels, unless asked for big endian.

<https://www.emva.org/wp-content/uploads/GenICam_SFNC_2_0_0.pdf>

```
12.2.66 GevSCPSBigEndian

Endianess of multi-byte pixel data for this stream.
This is an optional feature. A device that does not support this feature must support little-endian
and always leave that bit clear.
```

On my device, I have tried to set pgrPixelBigEndian to false and forced GevSCPSBigEndian , but without any effect on what is sent by the camera.

I'm closing this issue, as it is not aravis fault. Please reopen it if you don't think so.

closed 2024-08-25 by EmmanuelP

berke, 2024-08-26:
> Hi,
> 
> In the blackfly documentation, there is a `PixelBigEndian` feature. Isn't this feature set to False by default ?

Thanks for checking on your side.  Yes when I turn the device on I get `pgrPixelBigEndian = false` and as I said it has no effect.

> On my device, I have tried to set pgrPixelBigEndian to false and forced GevSCPSBigEndian , but without any effect on what is sent by the camera.

> I'm closing this issue, as it is not aravis fault. Please reopen it if you don't think so.

I agree it's likely to be a PG FW bug, but is there a way of adding an option in aravis for dealing with such cameras?
Maybe a device-specific kludge based on a table of IDs like in the Linux kernel, a configuration option, or if not a GUI toggle?  (I tried adding one but I'm not familiar enough with Gtk code to do it efficicently.)  Should I open a feature request instead?

[image]

EmmanuelP, 2024-08-26:
> I agree it's likely to be a PG FW bug, but is there a way of adding an option in aravis for dealing with such cameras?
> Maybe a device-specific kludge based on a table of IDs like in the Linux [...]

A device specific kludge is the way to go, like what is done for the legacy endianness mechanism here: <https://github.com/AravisProject/aravis/blob/2ebaa8661761ea4bbc4df878aa67b4a9e1a9a3b9/src/arvgcport.c#L63-L74>

You may open a feature request, but a merge request would be even better...