AravisProject/aravis #926: Unable To Detect Imperx Cheetah C4180M Camera

bagoodwin, 2024-09-02:
I am unable to detect my Imperx C4180M camera through aravis. For instance:
```
$ arv-camera-test-0.8
Looking for the first available camera
No camera found: No supported device found
```
I can open the viewer fine and run the other commands, but none of them are detecting my camera which is currently being detected by lsusb. I can also see the camera as /sys/class/usbmisc/u3v0.

**Install**
- Follow install instructions at [https://aravisproject.github.io/aravis/aravis-stable/building.html](https://aravisproject.github.io/aravis/aravis-stable/building.html)
- added /usr/local/lib to ld.so.conf.d
- ldconfig

**System**
- Arch Linux with kernel version 6.10.7
- X86_64 CPU

**lsusb**
`lsusb -v`
```
Bus 002 Device 005: ID 2db2:0200 Imperx Inc. U3V-C4180M-RF000
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 [unknown]
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x2db2 Imperx Inc.
  idProduct          0x0200 U3V-C4180M-RF000
  bcdDevice            0.00
  iManufacturer           1 Imperx Inc.
  iProduct                2 U3V-C4180M-RF000
  iSerial                 4 880409
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0074
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              400mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass        239 Miscellaneous Device
      bFunctionSubClass       5 USB3 Vision
      bFunctionProtocol       0 
      iFunction               3 USB3 Vision Device
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      0 
      iInterface             10 Device Control IF
      ** UNRECOGNIZED:  14 24 01 00 00 01 00 00 00 01 00 05 01 02 06 07 08 04 09 0c
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      1 
      iInterface             11 Device Event IF
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      2 
      iInterface             12 Device Streaming IF
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      HIRD Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000c
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   3
      Lowest fully-functional device speed is SuperSpeed (5Gbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat        2047 micro seconds
Device Status:     0x0000
  (Bus Powered)
```

EmmanuelP, 2024-09-02:
May be a permission issue ? Did you try to run the aravis tools as root ?

bagoodwin, 2024-09-02:
I don't remember if I tried that. I'll get back to you in about 2 days when I get a chance to test the camera again.

bagoodwin, 2024-09-03:
It was just a permission issue. My bad for not checking that before asking.

closed 2024-09-03 by bagoodwin