AravisProject/aravis #941: Packet socket mode and callback causes segfault on stream start

paulsammut, 2024-09-14:
**Describe the bug**
When running with Packet socket mode enabled I get a segfault on stream start.

**To Reproduce**
Run packet socket mode using a callback and get segfault on stream start.

**Expected behavior**
Run packet socket mode using a callback without segfault

**Camera description:**
 - Manufacturer: Lucid Vision
 - Model: PHX124S
 - Interface [e.g. USB3]: GigE

**Platform description:**
 - Aravis version: 0.8.30 hash: 96cea9835ce1535547d8680c8b0be33e97f6cb30
 - OS: [e.g. Fedora 20]: Fedora 40
 - Hardware [e.g. x86_64]: x86_64

**Additional context**
My code uses a callback to service frame buffers. I am setting priority realtime to 10, and enabling packet socket mode by running `sudo setcap cap_net_raw+ep`.

With the same code I do not get this error under the following conditions:
* Run in standard socket mode
* **Note** When I run with ARV_DEBUG logging enabled or in a Debug build mode this segfault becomes harder to trigger. I have to create a script to starts and stops the stream continuously for ~50 times before hitting the segfault. If I run without AVR_DEBUG logging and in an optimized build I get the segfault on every stream start.

The segfault is happening due to a null dereference of `frame` [in this line](https://github.com/AravisProject/aravis/blob/96cea9835ce1535547d8680c8b0be33e97f6cb30/src/arvgvstream.c#L384).
 
Stack trace of offending code:
```
#0  0x0000000003143899 in _find_frame_data (thread_data=0x7fb92067b000, packet=0x7fb94d6000ac, packet_size=62, frame_id=3759, packet_id=1580364, read_count=62, time_us=27579292116) at ./third-party/aravis/src/arvgvstream.c:384
#1  0x000000000314299c in _process_packet (thread_data=0x7fb92067b000, packet=0x7fb94d6000ac, packet_size=62, time_us=27579292116) at ./third-party/aravis/src/arvgvstream.c:899
#2  0x0000000003141af2 in _ring_buffer_loop (thread_data=0x7fb92067b000) at ./third-party/aravis/src/arvgvstream.c:1336
#3  0x00000000031414d8 in arv_gv_stream_thread (data=0x7fb92067b000) at ./third-party/aravis/src/arvgvstream.c:1383
#4  0x000000000341bbf7 in g_thread_proxy (data=0x7fb920608940) at gthread.c:784
#5  0x00007fb9a68b96d7 in start_thread () from /lib64/libc.so.6
#6  0x00007fb9a693d60c in clone3 () from /lib64/libc.so.6
```

Info on locals showing frame being null:
```
(gdb) info locals
frame = 0x0
buffer = 0x7fb8b40ec770
iter = 0x0
n_packets = 0
frame_id_inc = 1
extended_ids = 0
```

Output before segfault:
```
[INFO] [1726290134.182623044] [cam1.genicam]: Creating stream in id: 19968
[22:02:14.184] 🅸 device> [GvDevice::create_stream] Number of stream channels = 1
[22:02:14.184] 🅸 stream> [GvStream::stream_new] Stream channel = 0
[22:02:14.189] 🅸 stream> [GvStream::stream_new] Packet size = 8999 byte(s)
[22:02:14.195] 🅸 stream> [GvStream::stream_new] Destination stream port = 57561
[22:02:14.195] 🅸 stream> [GvStream::stream_new] Source stream port = 49154
[INFO] [1726290134.195879588] [cam1.genicam]: Stream cb called in 25956
[INFO] [1726290134.196070041] [cam1.genicam]: Set stream thread realtime priority to 10
[22:02:14.205] 🅸 stream> [GvStream::loop] Packet socket method
[22:02:14.249] 🅸 stream-thread> [GvStream::set_socket_filter] source ip = 0xc0a8fb65 - port = 49154 - dest ip = 0xc0a8fb0b - port 57561
[INFO] [1726290134.250391460] [cam1.genicam]: Enabling performance improvements for GigE Vision cameras
[INFO] [1726290134.252588018] [cam1.genicam]: Created 32 buffers of size 16200000 bytes
Segmentation fault (core dumped)
```

aldoshkind, 2024-11-21:
Hi!
Faced same problem on version 0.9.0.
Running from docker. Camera is `Lucid Vision Labs-TRI028S-M`

closed 2024-11-27 by EmmanuelP

paulsammut, 2024-12-13:
tested this fix and it works thank you very much!

PR #959 (2024-11-27): gvstream: fix wrong access to not yet initialized frame data
The buffer we are processing here is not yet stored in frame data. Fix #941
commit c07e8be6 (2024-11-27): gvstream: fix wrong access to not yet initialized frame data

The buffer we are processing here is not yet stored in frame data.
Fix #941
commit cad2a83a (2024-11-27): gvstream: fix wrong access to not yet initialized frame data

The buffer we are processing here is not yet stored in frame data.
Fix #941