AravisProject/aravis #942: Installed headers don't play nice with CMake

themightyoarfish, 2024-09-23:
**Describe the bug**

Aravis headers include other aravis headers with e.g. ` #include <arvtypes.h>`, without an intermediate ´aravis/` subdirectory, which is created on installation. This means that client projects need to not only include `<prefix>/include` as an include directory, but also `<prefix>/include/aravis`. This lack of namespacing opens the door for file name collisions. Admittedly this is currently unlikely, since all headers start with `arv…`, but in my opinion this is still unconventional. I've encountered this problem before, where some library came with an `event.h` header which shadowed a macOS system header.

 It's more usual to either include everything as `#include <aravis/…>` for build and install tree (necessitating moving all headers into a separate `aravis/` directory in the build), or to use double quotes `#include "arvtypes.h"` in aravis headers.

**To Reproduce**

Install aravis, add just `<prefix>/include` do include path, try to build something that includes `<aravis/arv.h>`.

**Expected behavior**
 
Including `<prefix>/include` as an include path is sufficient to build against aravis.

**Camera description:**
n/a

**Platform description:**
 - 0.8
 - OS: ubuntu 20.04
 - x86_64

**Additional context**

It seems to me that just using double quotes would be the simplest solution.

themightyoarfish, 2024-09-23:
Its also a bit strange that the default header install dir is `include-<version>` instead of just `include`, although i can see the rationale here. More odd i think is the additional subdir for libraries `<prefix>/lib/aravis`, which probably won't be in any default linker search path, will it? First time I see a library do this, maybe I'm missing something here.

EmmanuelP, 2024-09-30:
Unless told otherwise, aravis should be installed in `$prefix`, with the following directories:

```
$prefix/bin
$prefix/include/aravis-0.8/
$prefix/lib64/
$prefix/lib64/pkgconfig/
$prefix/share/gir-1.0/
```
Libraries and applications are suffixed with the API version. Any clash with other system installed software is unlikely.

Please attach the console output of `ninja install`.