AravisProject/aravis #943: Namespace Aravis not available

drmcnelson, 2024-09-27:
**Describe the bug**
A clear and concise description of what the bug is.

Fresh install of aravis, per build instructions, host is Fedora 40.

Running a Python program that has previously worked on other platforms, the following error text is produced:

Traceback (most recent call last):
  File "/home/datacollection/TeensyDataAcquistion/Python_Programs/FireFly.py", line 48, in <module>
    gi.require_version ('Aravis', '0.8')
  File "/usr/lib64/python3.12/site-packages/gi/__init__.py", line 122, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace Aravis not available

**Camera description:**
 - Manufacturer  FLIR
 - Model  Blackfly BFLY-U3-13S2M
 - Interface [e.g. USB3]  USB3

**Platform description:**
 - Aravis version
 - OS: [e.g. Fedora 20]  Fedora 40
 - Hardware [e.g. x86_64]   x86_64

**Additional context**
Add any other context about the problem here.

drmcnelson, 2024-09-27:
Same as issue from a few years ago,  https://github.com/AravisProject/aravis/issues/803#issue-1797787801

That cures the first bug, still have a bug not finding a run time library.

drmcnelson, 2024-09-27:
Next issue,  the computer does not boot while the camera is connected.

EmmanuelP, 2024-09-30:
I'm a bit confused about the exact issue. I guess it is only related to the gobject introspection based python bindings, which you have solved by yourself. Please open another issue if needed?

closed 2024-09-30 by EmmanuelP

drmcnelson, 2024-09-30:
Well the issue in this post is the name space, it requires a line added to the bashrc.

The library issue; we just talked about it in another thread.

Both of those are really documentation issues.     We rediscover them every few years when we install on a new machine.  So, a little more detail in the documentation might cure that for good.

The third issue:

When this particular camera is connected, the computer hangs in boot.   When the camera is disconnected the computer continues through the boot and everything is okay.    After the boot completes we can connect the camera and use it with the aravis software with no issues.

All of that is after installing aravis.  I think it did boot okay before aravis was installed, by I am not sure.