AravisProject/aravis #944: Fails to find libaravis-0.8.s0.0

drmcnelson, 2024-09-27:
**Describe the bug**
A clear and concise description of what the bug is.

** (process:2521): WARNING **: 14:46:56.440: Failed to load shared library 'libaravis-0.8.so.0' referenced by the typelib: libaravis-0.8.so.0: cannot open shared object file: No such file or directory
Traceback (most recent call last):
  File "/home/datacollection/TeensyDataAcquistion/Python_Programs/FireFly.py", line 1538, in <module>
    ids = get_device_ids()
          ^^^^^^^^^^^^^^^^
  File "/home/datacollection/TeensyDataAcquistion/Python_Programs/FireFly.py", line 113, in get_device_ids
    Aravis.update_device_list()
gi.repository.GLib.GError: g-invoke-error-quark: Could not locate arv_update_device_list: libaravis-0.8.so.0: cannot open shared object file: No such file or directory (1)

Aravis 0.8.33

Fedora 40

Blackfly BFLY-U3-13S2M

drmcnelson, 2024-09-27:
Solution:

Using find / -iname libaravis-0.8.so.0,  we find

    /usr/local/lib64/libaravis-0.8.so.0

Therefore, add the following file to the system

  /etc/ld.so.conf.d/aravis.conf

And this file should have a single line:

   /usr/local/lib64/

And then run

   # sudo ldconfig

Note this is an installation bug.   The install should be creating this file, at a minimum, and consider also running ldconfig or prompting the user to reboot.

EmmanuelP, 2024-09-30:
Running ldconfig by default is no a good idea: <https://github.com/mesonbuild/meson/pull/9926>

closed 2024-09-30 by EmmanuelP

drmcnelson, 2024-09-30:
Agreed, it might not always be a good idea, and therefore it is not a good idea. 

Advising the user to reboot at the end of the install could be a good idea.

Also,  since it is a recurring issue, perhaps  the install could be a little more automatic or the instructions could be a little more complete.

BTW, the package is fantastic one it is installed and running.  I have used it for a gated and timer series hyper spectral imaging, so aravis really is enabling for my work.  Thank you for doing it.