AravisProject/aravis #945: Blackfly BFLY-U3-13S2M,  not recognized

drmcnelson, 2024-09-27:
**Describe the bug**
A clear and concise description of what the bug is.

Blackfly Blackfly BFLY-U3-13S2M attached to computer.

Not recognized by Aravis.

Aravis 0.8.33
Fedora 40
Blackfly BFLY-U3-13S2M

$ lsusb
Bus 001 Device 008: ID 1e10:3300 Point Grey Research, Inc. Blackfly BFLY-U3-13S2M

datacollection@rothberglab01:~$ FireFly.py 
/home/datacollection/TeensyDataAcquistion/Python_Programs/ImageWindow.py:31: UserWarning: Treat the new Tool classes introduced in v1.5 as experimental for now; the API and rcParam may change in future versions.
  plt.rcParams['toolbar'] = 'toolmanager'
[]
no cameras available
Traceback (most recent call last):
  File "/home/datacollection/TeensyDataAcquistion/Python_Programs/FireFly.py", line 1545, in <module>
    camera = FireFlyCamera( ids[args.devicenumber],
                            ~~~^^^^^^^^^^^^^^^^^^^
IndexError: list index out of range

EmmanuelP, 2024-09-30:
Hi,

You probably have not installed the udev file that allows any user to connect to the USB3Vision device. Please reopen if it is another issue.

closed 2024-09-30 by EmmanuelP

drmcnelson, 2024-09-30:
Thank you, and indeed you are right.  Where do I find the udev file? 

Perhaps it should be mentioned in the installation instructions along with a link to the file?

EmmanuelP, 2024-09-30:
<https://aravisproject.github.io/aravis/aravis-stable/usb.html>