AravisProject/aravis #946: problem with alied vision gige camera

james-ht-tech, 2024-10-07:
**Describe the bug**
Hi i installed Aravis 0.8 according to installation(meson) and everything installed fine but when i run `/usr/local/bin# ./arv-tool-0.8` i encountered `No device found`  ( i set my ip and set mtu to 9000) , but  i have not any problem with this Genicam in windows OS by its official app  by setting ip and enabled jambo frame i dont know why?  

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Camera description:**
 - AliedVision
 - Model
 - Gige

**Platform description:**
 - Aravis version
 - OS: [e.g. Fedora 20]
 - Jetson Orin(arm64)

**Additional context**
Add any other context about the problem here.

EmmanuelP, 2024-10-07:
How is the camera connected to the machine ?

james-ht-tech, 2024-10-14:
it is Gige camera and connected via RJ45 cable

EmmanuelP, 2024-10-25:
Firewall ?

What is the output of `arv-tool-0.8 --debug all` ?