AravisProject/aravis #947: Cognex IS8400 GigE camera not detected!

shrikadam, 2024-10-07:
**Running ```arv-tool-0.8 -d all``` gives following output:**
```console
[14:01:02.472] 🅸 interface> Found 0 USB3Vision device (among 7 USB devices)
[14:01:02.473] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[14:01:02.473] 🅸 interface> [GvDiscoverSocket::new] Add interface 169.254.0.0 (169.254.255.255)
[14:01:02.473] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.146.103 (192.168.146.255)
[14:01:02.473] 🅸 interface> [GvDiscoverSocket::new] Add interface 172.18.0.1 (172.18.255.255)
[14:01:02.473] 🅸 interface> [GvDiscoverSocket::new] Add interface 172.17.0.1 (172.17.255.255)
No device found
```

**I am able to ping to the camera with ```ping 169.254.0.1```:**
```
PING 169.254.0.1 (169.254.0.1): 56 data bytes
64 bytes from 169.254.0.1: icmp_seq=0 ttl=255 time=0.692 ms
64 bytes from 169.254.0.1: icmp_seq=1 ttl=255 time=0.411 ms
64 bytes from 169.254.0.1: icmp_seq=2 ttl=255 time=0.434 ms
64 bytes from 169.254.0.1: icmp_seq=3 ttl=255 time=0.410 ms
```

**I have verified that the firewall is off with ```sudo ufw status```:**
```console
Status: inactive
```

**Camera description:**
 - Manufacturer: Cognex
 - Model: IS8400M
 - Interface: GigE

**Platform description:**
 - Aravis 0.8
 - OS: Ubuntu 20.04
 - Hardware: x86_64

EmmanuelP, 2024-10-07:
How the camera is connected to the machine ?

shrikadam, 2024-10-07:
Camera has an M12 connector with which it is connected to PC for data transfer and to PSU for 48V power using a breakout cable. With Cognex' proprietory software (InSight Explorer), I am able to at least detect the camera and get a live feed (although that too did not detect the sensor by default; I had to manually scan the peripherals). Let me know what you think; I'll try to add an image of the setup here tomorrow.

Thank you for the reply and very much grateful for this project!

EmmanuelP, 2024-10-25:
I don't know what is causing your issue. What could help is a wireshark capture of the traffic on the 169.254.0.0 interface while aravis is trying to connect to the camera.

closed 2025-06-23 by shrikadam