AravisProject/aravis #952: BAYER10 format is not supported

Rz-Rz, 2024-10-23:
**Is your feature request related to a problem? Please describe.**
My camera is outputting a BAYER10 streams but aravis detects it as a mono10. 

**Describe the solution you'd like**
It would be nice if it could correctly parse it, in the viewer etc. It would be nice to be able to simulate a BAYER10 streams aswell using the fake camera.

**Describe alternatives you've considered**
Using GStreamer to produce a BAYER10 streams to test with it?

EmmanuelP, 2024-10-25:
What does tell you aravis detects bayer10 as mono10 ?

Rz-Rz, 2024-10-25:
It's the viewer that automatically capture the stream as mono10, and I can't seem to set it as BAYER10

EmmanuelP, 2024-10-25:
Ah ok. That is because gstreamer does not support bayer10 (at least the last time I checked). arv-viewer picks the first supported pixel format. I'm not willing to add pixel fomat conversion in aravis, it is out of the scope of this library.

Cheers.

EmmanuelP, 2024-10-25:
That being said, the day gstreamer supports bayer10, I will happily add the support for it in arv-viewer.

closed 2024-10-25 by EmmanuelP