AravisProject/aravis #955: Fake camera with a mock video pattern don't get picked up by arvtools/arviewer.

Rz-Rz, 2024-10-25:
**Describe the bug**
I'm trying to build a fake camera source with my own specific pattern using opencv. However the camera starts but seems to never get picked up by the arvtools/arvviewer.

**To Reproduce**
Here's my code: 
```cpp
#include <arv.h>
#include <glib.h>
#include <iostream>
#include <opencv2/opencv.hpp>
#include <signal.h>

// Define image properties
#define WIDTH 320
#define HEIGHT 240
#define PIXEL_FORMAT ARV_PIXEL_FORMAT_MONO8 // Using MONO8 for simplicity

struct FillPatternData {
  guint64 frame_counter;
  int width;
  int height;
};

// Correct callback signature matching ArvFakeCameraFillPattern
void fill_pattern_callback(ArvBuffer *buffer, void *user_data, guint32 width,
                           guint32 height, guint32 padding_x) {
  FillPatternData *data = static_cast<FillPatternData *>(user_data);

  // Generate a new 8-bit grayscale image using OpenCV
  cv::Mat gray_image(data->height, data->width, CV_8UC1, cv::Scalar(0));

  // Create a dynamic pattern that changes with frame_counter
  // For simplicity, we'll create a moving horizontal gradient
  for (int y = 0; y < data->height; y++) {
    for (int x = 0; x < data->width; x++) {
      gray_image.at<uchar>(y, x) =
          static_cast<uchar>((x + data->frame_counter) % 256);
    }
  }

  // Get the buffer data pointer and size
  size_t buffer_size;
  void *buffer_data = (void *)arv_buffer_get_data(buffer, &buffer_size);
  if (!buffer_data) {
    std::cerr << "Failed to get buffer data" << std::endl;
    return;
  }

  // Ensure that the generated image fits into the buffer
  size_t image_size = data->width * data->height; // 1 byte per pixel for MONO8
  if (image_size > buffer_size) {
    std::cerr << "Image size exceeds buffer size. Image size: " << image_size
              << ", Buffer size: " << buffer_size << std::endl;
    return;
  }

  // Copy the grayscale image data into the buffer
  memcpy(buffer_data, gray_image.data, image_size);

  // Increment frame counter
  data->frame_counter++;
}

static gboolean stop = FALSE;

static void control_c_signal(int signal) { stop = TRUE; }

int main(int argc, char **argv) {
  // Initialize the type system (required by GObject in older GLib versions)
#if !GLIB_CHECK_VERSION(2, 35, 0)
  g_type_init();
#endif

  // Initialize Aravis
  arv_update_device_list();

  // Set the GenICam XML file for the fake camera
  arv_set_fake_camera_genicam_filename("fake_camera_genicam.xml");

  // Create a fake GigE Vision camera
  ArvFakeCamera *gv_fake_camera =
      arv_fake_camera_new_full("3", "fake_camera_genicam.xml");
  if (!gv_fake_camera) {
    std::cerr << "Failed to create fake camera" << std::endl;
    return EXIT_FAILURE;
  }

  // Initialize fill pattern data
  FillPatternData fill_data;
  fill_data.frame_counter = 0;
  fill_data.width = WIDTH;
  fill_data.height = HEIGHT;

  // Set fill pattern callback with the correct signature
  arv_fake_camera_set_fill_pattern(ARV_FAKE_CAMERA(gv_fake_camera),
                                   fill_pattern_callback, &fill_data);

  // Set trigger frequency (frame rate)
  arv_fake_camera_set_trigger_frequency(ARV_FAKE_CAMERA(gv_fake_camera),
                                        10.0); // 10 FPS

  std::cout << "Fake Aravis camera is running. Press Ctrl+C to stop."
            << std::endl;

  // Handle SIGINT and SIGTERM to allow graceful exit
  signal(SIGINT, control_c_signal);
  signal(SIGTERM, control_c_signal);

  // Create and run the main loop
  GMainLoop *main_loop = g_main_loop_new(NULL, FALSE);

  // Run the main loop until interrupted
  while (!stop) {
    g_main_context_iteration(NULL, FALSE);
  }

  // Cleanup
  g_main_loop_unref(main_loop);
  g_object_unref(gv_fake_camera);

  return EXIT_SUCCESS;
}
```

The xml context I used: 
```xml
<?xml version="1.0" encoding="UTF-8"?>
<GenICamXML>
  <Device>
    <DeviceModelName>Fake_Aravis_Camera</DeviceModelName>
    <DeviceVendorName>FakeVendor</DeviceVendorName>
    <SerialNumber>3</SerialNumber>
    
    <!-- Pixel Format Enumeration -->
    <PixelFormat>
        <EnumEntry name="Mono8">
    <Value>0x01080001</Value> <!-- Standard GenICam value for Mono8 -->
  </EnumEntry>
      <!-- You can add more pixel formats here if needed -->
    </PixelFormat>
    
    <!-- Width Parameter -->
    <Width>
      <Int64Min>1</Int64Min>
      <Int64Max>800</Int64Max>
      <Int64Inc>1</Int64Inc>
      <Int64Default>320</Int64Default>
    </Width>
    
    <!-- Height Parameter -->
    <Height>
      <Int64Min>1</Int64Min>
      <Int64Max>600</Int64Max>
      <Int64Inc>1</Int64Inc>
      <Int64Default>240</Int64Default>
    </Height>
    
    <!-- Trigger Frequency Parameter -->
    <TriggerFrequency>
      <FloatMin>0.0</FloatMin>
      <FloatMax>1000.0</FloatMax>
      <FloatInc>0.1</FloatInc>
      <FloatDefault>10.0</FloatDefault>
    </TriggerFrequency>
    
    <!-- Acquisition Mode Enumeration -->
    <AcquisitionMode>
      <EnumEntry name="Continuous">
        <Value>0</Value>
      </EnumEntry>
      <EnumEntry name="SingleFrame">
        <Value>1</Value>
      </EnumEntry>
      <EnumEntry name="MultiFrame">
        <Value>2</Value>
      </EnumEntry>
    </AcquisitionMode>
    
    <!-- Add more parameters as needed -->
  </Device>
<GenICamXML/>
```

**Expected behavior**
The camera should appear in the arviewer or in the arvtools.

**Platform description:**
 - Aravis version 0.8.33
 - OS: 5.10.192-tegra
 - Hardware arm64

EmmanuelP, 2024-11-22:
Hi,

The XML data you have attached is really what you are giving to the fake camera ? It is highly incomplete.

Rz-Rz, 2025-02-25:
Do you have an example on how to format them ? Since I have limited knowledge of aravis/gigevision I would love to have any kind of info on how to properly populate the xml document.

Rz-Rz, 2025-02-25:
Using arv-tool-0.8 I got : 
                                                                                                                                                                        
                                                                                                                                                                                       
```bash
Aravis-Fake-GV01 (127.0.0.1)                                                                                                                                                           
Aravis-Fake-SIMCAM01 (127.0.0.1) 
```

But when reading them :
```bash
gst-launch-1.0 -v aravissrc camera-name=Fake_1 ! videoconvert ! autovideosink                                                                                                        
                                                                                                                                                                                       
Setting pipeline to PAUSED ...                                                                                                                                                         
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Could not find camera "Fake_1": Device 'Fake_1' not found                                                                
Additional debug info:                                                                                                                                                                 
../gst/gstaravis.c(457): gst_aravis_init_error (): /GstPipeline:pipeline0/GstAravis:aravis0                                                                                            
ERROR: pipeline doesn't want to preroll.                                                                                                                                               
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: GStreamer error: state change failed and some element failed to post a proper error message with the reason for the failu
re.                                                                                                                                                                                    
Additional debug info:                                                                                                                                                                 
../libs/gst/base/gstbasesrc.c(3556): gst_base_src_start (): /GstPipeline:pipeline0/GstAravis:aravis0:                                                                                  
Failed to start                                                                                                                                                                        
ERROR: pipeline doesn't want to preroll.                                                                                                                                               
Failed to set pipeline to PAUSED.                                                                                                                                                      
Setting pipeline to NULL ...                                                                                                                                                           
Freeing pipeline ...              

// OR

gst-launch-1.0 -v aravissrc camera-name=Aravis-Fake-SIMCAM01 ! videoconvert ! autovideosink                                                                                          
                                                                                                                                                                                       
Setting pipeline to PAUSED ...                                                                                                                                                         
Pipeline is live and does not need PREROLL ...                                                                                                                                         
Pipeline is PREROLLED ...                                                                                                                                                              
Setting pipeline to PLAYING ...                                                                                                                                                        
New clock: GstSystemClock                                                                                                                                                              
/GstPipeline:pipeline0/GstAravis:aravis0: stream = (null)                                                                                                                              
/GstPipeline:pipeline0/GstAravis:aravis0.GstPad:src: caps = video/x-raw, format=(string)GRAY8, width=(int)512, height=(int)512, framerate=(fraction)5/1                                
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:src: caps = video/x-raw, width=(int)512, height=(int)512, framerate=(fraction)5/1, format=(string)YUY2                     
/GstPipeline:pipeline0/GstAutoVideoSink:autovideosink0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-raw, width=(int)512, height=(int)512, framerate=(fraction)5/1, format=(st
[... cut ...]
Interrupt: Stopping pipeline ...                                                                                                                                                       
Execution ended after 0:00:05.321237118                                                                                                                                                
Setting pipeline to NULL ...                                                                                                                                                           
Freeing pipeline ...                                 
```

I just get black screen.

romainreignier, 2025-03-19:
Hi @Rz-Rz 
I am also trying to get a fake GV camera to output custom images.
I am struggling on copying data to the buffer.
In the `fill_pattern_callback`, `arv_buffer_get_data(buffer, &buffer_size);` returns a `buffer_size` of 0.

But about your problem, by copying the [arvfakegvcamera.c](https://github.com/AravisProject/aravis/blob/main/src/arvfakegvcamera.c) file, and removing the CLI args handling, I can confirm that I have a working fake camera and I am able to see the moving diagonal rainbow (default pattern) with this code:

```c
#include <arv.h>

#include <stdlib.h>
#include <stdio.h>

static gboolean cancel = FALSE;

static void
set_cancel (int signal)
{
    (void)signal;
    cancel = TRUE;
}

int
main (int argc, char **argv)
{
    ArvGvFakeCamera *gv_camera = arv_gv_fake_camera_new_full (NULL, NULL, NULL);

    signal (SIGINT, set_cancel);

    if (arv_gv_fake_camera_is_running (gv_camera))
        while (!cancel)
            g_usleep (1000000);
    else
        printf ("Failed to start camera\n");

    g_object_unref (gv_camera);

    return EXIT_SUCCESS;
}
```

As you see, no need to use a g_main_loop.

Rz-Rz, 2025-03-19:
It would be really great if we could choose the width/height, pixel pattern and feed the buffers.

romainreignier, 2025-03-19:
Yes, my sample was just for you to validate that you can run a basic fake camera.

The in-tree [arvfakecamera.c](https://github.com/AravisProject/aravis/blob/e7e3560095ba5dc6834892f997fbeb3ef4d4e327/src/arvfakecamera.c) uses the private interface of the buffer:

```c
buffer->priv->data
```

And sets the variable `received_size` for the data size:
https://github.com/AravisProject/aravis/blob/e7e3560095ba5dc6834892f997fbeb3ef4d4e327/src/arvfakecamera.c#L546

But these API are not available from an installed library.

We only have [`arv_buffer_get_data()`](https://aravisproject.github.io/aravis/aravis-stable/method.Buffer.get_data.html) but that should be used to access the buffer data and does not modify the `received_size` variable.

I have used Github search to find some code using `arv_fake_camera_set_fill_pattern()` to get a working example.
https://github.com/search?q=arv_fake_camera_set_fill_pattern+-path%3Aarvfakecamera.c+-path%3Aarvfakecamera.h+-path%3Afake.c+-is%3Afork+language%3AC&type=code&l=C
But the only I could find is this one:
https://github.com/thodnev/gigevision-xilinx/blob/0dca67d9f1abe80fd5a327211517419ccce6e7c3/ArvFakeModule/arvfakegvcamera.c#L1076
In which they use `#include "arvbufferprivate.h"`.

Maybe @EmmanuelP can help us on how to provide custom images to a fake camera outside of the aravis source tree?

romainreignier, 2025-03-19:
By looking closer at the linked repo, they are including the aravis source code in their [Makefile](https://github.com/thodnev/gigevision-xilinx/blob/0dca67d9f1abe80fd5a327211517419ccce6e7c3/ArvFakeModule/Makefile#L12C1-L12C31)

Doing the same or copying the three files inside my project:
* `arvbufferprivate.h`
* `arvgvspprivate.h`
* `arvdebugprivate.h`

And with this (not very clean) code:

```c
/* Aravis - Digital camera library
 *
 * Copyright © 2009-2022 Emmanuel Pacaud
 *
 * This library is free software; you can redistribute it and/or
 * modify it under the terms of the GNU Lesser General Public
 * License as published by the Free Software Foundation; either
 * version 2 of the License, or (at your option) any later version.
 *
 * This library is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 * Lesser General Public License for more details.
 *
 * You should have received a copy of the GNU Lesser General
 * Public License along with this library; if not, write to the
 * Free Software Foundation, Inc., 51 Franklin Street, Fifth Floor,
 * Boston, MA 02110-1301, USA.
 *
 * Author: Emmanuel Pacaud <emmanuel.pacaud@free.fr>
 */

#include <arv.h>
#include <arvbuffer.h>

#include "arvbufferprivate.h"

#include <assert.h>
#include <stdio.h>
#include <stdlib.h>

static gboolean cancel = FALSE;

static void set_cancel(int signal)
{
    (void)signal;
    cancel = TRUE;
}

static char* arv_option_interface_name = NULL;
static char* arv_option_serial_number = NULL;
static char* arv_option_genicam_file = NULL;
static char* arv_option_image_file = NULL;
static double arv_option_gvsp_lost_ratio = 0.0;
static char* arv_option_debug_domains = NULL;

static const GOptionEntry arv_option_entries[] = {
    {"interface", 'i', 0, G_OPTION_ARG_STRING, &arv_option_interface_name,
     "Listening interface name or address", "interface"},
    {"serial", 's', 0, G_OPTION_ARG_STRING, &arv_option_serial_number,
     "Fake camera serial number", "serial_nbr"},
    {"genicam", 'g', 0, G_OPTION_ARG_STRING, &arv_option_genicam_file,
     "XML Genicam file to use", "genicam_filename"},
    {"image", 'b', 0, G_OPTION_ARG_STRING, &arv_option_image_file,
     "Raw image file to use", "image_filename"},
    {"gvsp-lost-ratio", 'r', 0, G_OPTION_ARG_DOUBLE, &arv_option_gvsp_lost_ratio,
     "GVSP lost packet ratio", "packet_per_thousand"},
    {"debug", 'd', 0, G_OPTION_ARG_STRING, &arv_option_debug_domains, NULL,
     "{<category>[:<level>][,...]}"},
    {NULL}};
[... 107 lines cut ...]
    signal(SIGINT, set_cancel);

    if(arv_gv_fake_camera_is_running(gv_camera))
        while(!cancel) g_usleep(1000000);
    else
        printf("Failed to start camera\n");

    g_object_unref(gv_camera);

    if(buffer)
    {
        free(buffer);
    }

    return EXIT_SUCCESS;
}
```

Calling it with:

```
./aravis_fake_gv_real_images -b my_image.raw
```

I have my image displayed in the arv-viewer 🚀