AravisProject/aravis #956: Aravis + Python + Docker takes no pictures

Fran-ATRIA, 2024-11-05:
I’m just starting to use Aravis and I’m stuck trying to get the Python libary to work.

My system is a Docker (ubuntu:latest) container running over Windows 10 with WSL2. I have a HIKVision GigE camera connected on my local network and I can see it from inside the container (via ping).

I believe everything is properly installed, because from my init code:
```python
self.cam = Aravis.Camera.new(self.ip)
self.stream = self.cam.create_stream(None, None)
self.device = self.cam.get_device()
print(f"Camera vendor : {self.cam.get_vendor_name ()}")
print(f"Camera model  : {self.cam.get_model_name ()}")
print(f"Pixel format  : {self.cam.get_pixel_format_as_string()}")
print(f"Stream: {self.stream}")
print(f"Device: {self.device}")
```

I can get these variables:
```sh
Camera vendor : Hikrobot
Camera model  : MV-CS200-10GC
Pixel format  : BayerGB8
Stream: <Aravis.GvStream object at 0x7f2025103c80 (ArvGvStream at 0x25c11a0)>
Device: <Aravis.GvDevice object at 0x7f2025109680 (ArvGvDevice at 0x1f49830)>
```

When I launch my code, this fragment:
```python
payload = self.cam.get_payload()
self.stream.push_buffer(Aravis.Buffer.new_allocate(payload))
self.cam.start_acquisition()
image = self.stream.timeout_pop_buffer(10000000)
data = image.get_data()
```

Gives the following error:
`AttributeError: 'NoneType' object has no attribute 'get_data'`

If I use a non-timeout function, I’m stuck forever waiting for the buffer to pop.

What might be going on here? Can you please help me?

Thanks in advance.

Fran-ATRIA, 2024-11-05:
Also some extra info, this is what I see on the debug:

```python
[11:53:07.556] 🅸 stream > [GvStream::stream_new] Stream channel = 0
[11:53:07.561] 🅸 stream > [GvStream::stream_new] Packet size = 8164 byte(s)
[11:53:07.586] 🅸 stream > [GvStream::stream_new] Destination stream port = 57851
[11:53:07.586] 🅸 stream > [GvStream::stream_new] Source stream port = 9001
[11:53:07.657] 🅸 stream > [GvStream::loop] Packet socket method
[11:53:08.879] 🅸 stream >     bins    ;frame_retent; packet_time;inter_packet
                                    0;           0;           0;           0
                         -------------
                         >=      2000;           0;           0;           0
                         <          0;           0;           0;           0
                         min         :         n/a;         n/a;         n/a
                         max         :         n/a;         n/a;         n/a
                         last max at :           0;           0;           0
                         counter     :           0:           0:           0
[11:53:08.879] 🅸 stream > [GvStream::finalize] n_completed_buffers    = 0
[11:53:08.879] 🅸 stream > [GvStream::finalize] n_failures             = 0
[11:53:08.879] 🅸 stream > [GvStream::finalize] n_underruns            = 0
[11:53:08.879] 🅸 stream > [GvStream::finalize] n_timeouts             = 0
[11:53:08.879] 🅸 stream > [GvStream::finalize] n_aborted              = 0
[11:53:08.880] 🅸 stream > [GvStream::finalize] n_missing_frames       = 0
[11:53:08.880] 🅸 stream > [GvStream::finalize] n_size_mismatch_errors = 0
[11:53:08.880] 🅸 stream > [GvStream::finalize] n_received_packets     = 0
[11:53:08.880] 🅸 stream > [GvStream::finalize] n_missing_packets      = 0
[11:53:08.880] 🅸 stream > [GvStream::finalize] n_error_packets        = 0
[11:53:08.880] 🅸 stream > [GvStream::finalize] n_ignored_packets      = 0
[11:53:08.880] 🅸 stream > [GvStream::finalize] n_resend_requests      = 0
[11:53:08.880] 🅸 stream > [GvStream::finalize] n_resent_packets       = 0
[11:53:08.880] 🅸 stream > [GvStream::finalize] n_resend_ratio_reached = 0
[11:53:08.880] 🅸 stream > [GvStream::finalize] n_resend_disabled      = 0
[11:53:08.880] 🅸 stream > [GvStream::finalize] n_duplicated_packets   = 0
[11:53:08.880] 🅸 stream > [GvStream::finalize] n_transferred_bytes    = 0
[11:53:08.880] 🅸 stream > [GvStream::finalize] n_ignored_bytes        = 0
[11:53:08.880] 🅸 stream > [Stream::finalize] Flush 1 buffer[s] in input queue
[11:53:08.880] 🅸 stream > [Stream::finalize] Flush 0 buffer[s] in output queue
```

Maybe it's related to [759](https://github.com/AravisProject/aravis/issues/759)?

Fran-ATRIA, 2024-11-12:
Thanks @eudoxos. But when you say native Windows build I understand that you mean Docker is not an option then?

KNSd2, 2024-11-12:
> Also some extra info, this is what I see on the debug:
>
> ```python
> [11:53:07.556] 🅸 stream > [GvStream::stream_new] Stream channel = 0
> [11:53:07.561] 🅸 stream > [GvStream::stream_new] Packet size = [...]

Did you check if indows firewall blocks some packets? With the default firewall settings of Windows you can connect to the camera, but the stream is blocked (on my system anyway).

Fran-ATRIA, 2024-11-12:
@KNSd2 if I turn off the firewall and the antivirus the same happens, so I guess it's not that.

eudoxos, 2024-11-12:
This has nothing to do with firewall. GigEVision has two network channels (port connections), one for control and one for streaming. The streaming connection is initiated by the camera to an open port (this is what Aravis does) on the host (computer), but Aravis will open the port bound to the interface (IP address) seen from within the Docker, but that one is different the one than what the OS uses outside of Docker (with --net=host, they are the same, that's why Docker under Linux will work). See e.g. https://github.com/docker/for-win/issues/6736 .