AravisProject/aravis #966: how to execute arv-viewer ?

creativethinkb, 2024-12-11:
I built Aravis 0.10 on my Windows 10 PC through MSYS2 (MINGW64). After that, I executed both executable files in separate MSYS2 windows:

./arv-fake-gv-camera-0.10.exe
-> No errors appeared.

./arv-viewer-0.10.exe
-> An error appeared.

Errors:

(arv-tool-0.10.exe:57708): CRITICAL: 11:37:40.761: arv_gentl_interface_device_infos_new: assertion 'serial_nbr != NULL' failed

(arv-tool-0.10.exe:57708): CRITICAL: 11:37:40.762: arv_gentl_interface_device_infos_ref: assertion 'infos != NULL' failed

Segmentation fault

creativethinkb, 2024-12-11:
Additionally, I’m wondering if I need to provide specific parameters when running the executable files. Could you advise what parameters need to be passed to the executables for proper execution?

EmmanuelP, 2025-01-22:
There is no specific parameter to set.

Please use the latest stable release, not the development version. And may be test the prebuilt packages: <https://packages.msys2.org/packages/mingw-w64-x86_64-aravis>.

closed 2025-01-22 by EmmanuelP