AravisProject/aravis #969: GStreamer-Video-CRITICAL **: 11:51:30.479: gst_video_center_rect: assertion 'src->h != 0' failed

VBUSSI, 2024-12-19:
**Describe the bug**
gst_video_center_rect: assertion 'src->h != 0'  error with arv-viewer

**To Reproduce**
Run arv-viewer.
Camera is detected but, no image returns when I start grab. I get the following error : GStreamer-Video-CRITICAL **: 11:51:30.479: gst_video_center_rect: assertion 'src->h != 0' failed.

./arv-camera-test-0.10
Looking for the first available camera
vendor name            = Photonfocus AG
model name             = MV4-D1280U-H01-GT
device serial number   = 054000062125
image width            = 640
image height           = 480
horizontal binning     = 1
vertical binning       = 1
exposure               = 5000 µs
gain                   = 1 dB
payload                = 307200 bytes
gv n_stream channels   = 1
gv current channel     = 0
gv packet delay        = 0 ns
gv packet size         = 2084 bytes
  0 frame/s  -       0 MiB/s - 45 errors
  0 frame/s  -       0 MiB/s - 46 errors
  0 frame/s  -       0 MiB/s - 49 errors
^C  0 frame/s  -       0 MiB/s - 48 errors
n_completed_buffers    = 0
n_failures             = 188
n_underruns            = 1306
n_timeouts             = 188
n_aborted              = 0
n_missing_frames       = 38
n_size_mismatch_errors = 0
n_received_packets     = 1600
n_missing_packets      = 28475
n_error_packets        = 0
n_ignored_packets      = 1306
n_resend_requests      = 0
n_resent_packets       = 0
n_resend_ratio_reached = 190
n_resend_disabled      = 0
n_duplicated_packets   = 0
n_transferred_bytes    = 7560
n_ignored_bytes        = 40440

**Expected behavior**
Displayed video with arv-viewer. It works on Debian 10.

**Camera description:**
 - Manufacturer : Photonfocus AG
 - Model : MV4-D1280U-H01-GT
 - Interface [Ethernet]

**Platform description:**
 - Aravis version 0.10
 - OS: [AlmaLinux 9.5 / kernel 5.14]
 - Hardware [e.g. x86_64]

**Additional context**
Add any other context about the problem here.

EmmanuelP, 2025-01-22:
> GStreamer-Video-CRITICAL **: 11:51:30.479: gst_video_center_rect: assertion 'src->h != 0' failed.

You can ignore this error, due to the fact no valid image is ever received (n_completed_buffers = 0)

> It works on Debian 10.

On the same machine ?

EmmanuelP, 2025-01-22:
Does `arv-camera-test-0.8` work on the Alma 9 machine ?

VBUSSI, 2025-01-24:
Hello Emmanuel,

It works on almalinux.
The issue was due to the proxy being enabled and the MTU not being properly configured.

Best regards

EmmanuelP, 2025-01-24:
Thanks for the answer.

Cheers.

closed 2025-01-24 by EmmanuelP