AravisProject/aravis #973: no image recevie when using in software trigger mode at opt-MCC050-GGO camera

liyu9202, 2025-01-19:
**Describe the bug**
no image recevie when using in software trigger mode

**To Reproduce**
I receives images with ArvStream based on the sample code - arvexample.c  and issues #628 and I try to change the trigger source - software in the below codes:
    ArvDevice* dev = arv_camera_get_device(camera);
    arv_device_set_string_feature_value(
        dev, "AcquisitionMode", "Continuous", nullptr);

    arv_device_set_string_feature_value(
        dev, "TriggerSelector", "AcquisitionStart", nullptr);
    std::cout << arv_device_get_string_feature_value(
        dev, "TriggerSelector", nullptr)
        << std::endl;
    
    arv_device_set_string_feature_value(dev, "TriggerMode", "Off", nullptr);
    std::cout << arv_device_get_string_feature_value(dev, "TriggerMode", nullptr)
        << std::endl;

    arv_device_set_string_feature_value(
        dev, "TriggerSelector", "FrameStart", nullptr);
    arv_device_set_string_feature_value(dev, "TriggerMode", "On", nullptr);
    arv_device_set_string_feature_value(
        dev, "TriggerSource", "Software", nullptr);

    std::cout << arv_device_get_string_feature_value(
        dev, "TriggerSelector", nullptr)
        << std::endl;
    std::cout << arv_device_get_string_feature_value(
        dev, "TriggerSource", nullptr)
        << std::endl;
    std::cout << arv_device_get_string_feature_value(dev, "TriggerMode", nullptr)
        << std::endl;
    gboolean result =arv_camera_is_enumeration_entry_available(camera, "TriggerActivation","RisingEdge", NULL);
    if(result)
        arv_camera_set_string(camera, "TriggerActivation", "RisingEdge", nullptr);
The output of features are fine, when TriggerMode set Off, but when TriggerMode set off, software_trigger doesn't get image .

[image]

**Expected behavior**
I call arv_camera_software_trigger or arv_device_execute_command(TriggerSoftware), it should comes image generally.

**Camera description:**
 - OPT
 -opt-MCC050-GGO 
 -GigE

**Platform description:**
 - Aravis 0.9.0
 -OS: Windows 10 64bit
 -Hardware Intel I7

**Additional context**
My code is

[ctest.zip](https://github.com/user-attachments/files/18467460/ctest.zip)

EmmanuelP, 2025-01-22:
> Aravis 0.9.0

Please use the latest stable release.

Please test your device using `arv-camera-test-0.8` with software trigger enabled and `-d all`, and report the console output here.

liyu9202, 2025-01-23:
C:\Users\yulsu\Downloads\aravis-0.8.31\install\bin>arv-test-0.8.exe -d all
[13:01:41.165] 馃吀 interface> Found 0 USB3Vision device (among 6 USB devices)
[13:01:41.170] 馃吀 interface> [GvDiscoverSocket::new] Add interface 192.168.0.10 (192.168.0.255)
[13:01:41.171] 馃吀 interface> [GvDiscoverSocket::new] Add interface 192.168.8.159 (192.168.8.255)
[13:01:41.171] 馃吀 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.255.255.255)
[13:01:41.174] 馃吀 interface> [GvInterface::discovery] Device 'OPT Machine Vision-OPT-MCC050-GG0-P15DB060033' found (interface 192.168.0.10) user_id '' - MAC '00:02:c4:37:1c:b9'
Found 1 device
[13:01:41.685] 馃吀 interface> Found 0 USB3Vision device (among 6 USB devices)
[13:01:41.685] 馃吀 device> [GvDevice::new] Interface address = 192.168.0.10
[13:01:41.685] 馃吀 device> [GvDevice::new] Device address = 192.168.0.166
[13:01:41.688] 馃吀 device> [GvDevice::load_genicam] xml url = 'Local:OPT_MCC050_GG0.zip;f0000000;11ea4' at 0x200
[13:01:41.688] 馃吀 device> [GvDevice::load_genicam] Xml address = 0xf0000000 - size = 0x11ea4 - OPT_MCC050_GG0.zip
[13:01:41.928] 馃吀 device> [GvDevice::load_genicam] Zipped xml data
[13:01:41.967] 馃吀 genicam> [Gc::set_default_node_data] Add 'DeviceID'
[13:01:41.967] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationLLA'
[13:01:41.967] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationDHCP'
[13:01:41.968] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationPersistentIP'
[13:01:41.968] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPAddress'
[13:01:41.968] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentSubnetMask'
[13:01:41.968] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentDefaultGateway'
[13:01:41.968] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevNumberOfNetworkInterfaces'
[13:01:41.968] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentIPAddress'
[13:01:41.969] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentSubnetMask'
[13:01:41.969] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentDefaultGateway'
[13:01:41.969] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelCount'
[13:01:41.969] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelSelector'
[13:01:41.969] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPAddrCalc'
[13:01:41.970] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevTimestampTickFrequency'
[13:01:41.970] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPHostPort'
[13:01:41.970] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSFireTestPacket'
[13:01:41.970] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSDoNotFragment'
[13:01:41.970] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSBigEndian'
[13:01:41.970] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSPacketSize'
[13:01:41.970] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPD'
[13:01:41.970] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCDA'
[13:01:41.972] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCSP'
[13:01:41.972] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCCFGMultipart'
[13:01:41.978] 馃吀 device> [GvDevice::new] Device endianness = big
[13:01:41.978] 馃吀 device> [GvDevice::new] Packet resend     = yes
[13:01:41.978] 馃吀 device> [GvDevice::new] Write memory      = yes
[13:01:41.978] 馃吀 device> [GvDevice::new] Legacy endianness handling = no
Testing 'OPT Machine Vision:OPT-MCC050-GG0'
Genicam:Load                        SUCCESS
Genicam:Schema                      SUCCESS Schema validation disabled
Properties:SensorSizeReadout        SUCCESS
Properties:SensorSizeCheck          IGNORED
Properties:GainAvailable            SUCCESS
Properties:GainReadout              SUCCESS
Properties:ExposureTimeAvailable    SUCCESS
Properties:ExposureTimeReadout      SUCCESS
[13:01:42.031] 馃吀 device> [GvDevice::create_stream] Number of stream channels = 1
[13:01:42.057] 馃吀 device> [GvDevice::auto_packet_size] Current packet size check successfull (8164 bytes)
[13:01:42.062] 馃吀 stream> [GvStream::stream_new] Stream channel = 0
[13:01:42.069] 馃吀 stream> [GvStream::stream_new] Packet size = 8164 byte(s)
[13:01:42.077] 馃吀 stream> [GvStream::stream_new] Destination stream port = 53544
[13:01:42.078] 馃吀 stream> [GvStream::stream_new] Source stream port = 20202
[13:01:42.078] 馃吀 stream> [GvStream::loop] Standard socket method
[13:01:43.070] 馃吀 stream>     bins    ;frame_retent; packet_time;inter_packet
                                    0;           0;         330;        6550
                                 2000;           0;         305;           0
                                 4000;           0;         303;           0
                                 6000;           0;         304;           0
                                 8000;           0;         307;           0
                                10000;           0;         306;           0
                                12000;           0;         301;           0
                                14000;           0;         306;           0
                                16000;           0;         306;           0
                                18000;           0;         304;           0
                                20000;           0;         305;           0
                                22000;           0;         305;           0
[... 170 lines cut ...]
[... cut ...]
[13:01:45.646] 馃吀 stream> [Stream::finalize] Flush 2 buffer[s] in input queue
[13:01:45.647] 馃吀 stream> [Stream::finalize] Flush 0 buffer[s] in output queue
MultipleAcquisitionB:BufferCheck    SUCCESS 10/10
MultipleAcquisitionB:FrameRate      FAILURE 3.97 Hz (expected:5.00 Hz)
Multipart:NoSupport                 SUCCESS
Chunks:NoSupport                    FAILURE
GigEVision:NNetworkInterfaces       SUCCESS 1
GigEVision:NStreamChannels          SUCCESS 1
GigEVision:StreamChannel            SUCCESS

EmmanuelP, 2025-01-23:
**arv-camera-test-0.8**

liyu9202, 2025-01-23:
C:\Users\yulsu\Downloads\aravis-0.8.33\install\bin>arv-camera-test-0.8.exe -d all
Looking for the first available camera
[18:28:00.173] 馃吀 interface> Found 0 USB3Vision device (among 5 USB devices)
[18:28:02.843] 馃吀 interface> [GvDiscoverSocket::new] Add interface 192.168.0.10 (192.168.0.255)
[18:28:02.844] 馃吀 interface> [GvDiscoverSocket::new] Add interface 192.168.8.159 (192.168.8.255)
[18:28:02.845] 馃吀 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.255.255.255)
[18:28:02.846] 馃吀 interface> [GvInterface::discovery] Device 'OPT Machine Vision-OPT-MCC050-GG0-P15DB060033' found (interface 192.168.0.10) user_id '' - MAC '00:02:c4:37:1c:b9'
[18:28:02.847] 馃吀 device> [GvDevice::new] Interface address = 192.168.0.10
[18:28:02.848] 馃吀 device> [GvDevice::new] Device address = 192.168.0.166
[18:28:02.849] 馃吀 device> [GvDevice::load_genicam] xml url = 'Local:OPT_MCC050_GG0.zip;f0000000;11ea4' at 0x200
[18:28:02.850] 馃吀 device> [GvDevice::load_genicam] Xml address = 0xf0000000 - size = 0x11ea4 - OPT_MCC050_GG0.zip
[18:28:02.951] 馃吀 device> [GvDevice::load_genicam] Zipped xml data
[18:28:02.994] 馃吀 genicam> [Gc::set_default_node_data] Add 'DeviceID'
[18:28:02.995] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationLLA'
[18:28:02.996] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationDHCP'
[18:28:02.996] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPConfigurationPersistentIP'
[18:28:02.997] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentIPAddress'
[18:28:02.998] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentSubnetMask'
[18:28:02.999] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevCurrentDefaultGateway'
[18:28:02.999] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevNumberOfNetworkInterfaces'
[18:28:03.000] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentIPAddress'
[18:28:03.001] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentSubnetMask'
[18:28:03.001] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevPersistentDefaultGateway'
[18:28:03.001] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelCount'
[18:28:03.002] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevStreamChannelSelector'
[18:28:03.002] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPAddrCalc'
[18:28:03.002] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevTimestampTickFrequency'
[18:28:03.003] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPHostPort'
[18:28:03.003] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSFireTestPacket'
[18:28:03.004] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSDoNotFragment'
[18:28:03.004] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSBigEndian'
[18:28:03.004] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPSPacketSize'
[18:28:03.005] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCPD'
[18:28:03.005] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCDA'
[18:28:03.005] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCSP'
[18:28:03.006] 馃吀 genicam> [Gc::set_default_node_data] Add 'ArvGevSCCFGMultipart'
[18:28:03.008] 馃吀 device> [GvDevice::new] Device endianness = big
[18:28:03.009] 馃吀 device> [GvDevice::new] Packet resend     = yes
[18:28:03.009] 馃吀 device> [GvDevice::new] Write memory      = yes
[18:28:03.009] 馃吀 device> [GvDevice::new] Legacy endianness handling = no
vendor name            = OPT Machine Vision
model name             = OPT-MCC050-GG0
device serial number   = P15DB060033
image width            = 2592
image height           = 2048
exposure               = 43000 碌s
gain                   = 1 dB
payload                = 5308416 bytes
gv n_stream channels   = 1
gv current channel     = 0
gv packet delay        = 0 ns
[18:28:03.039] 馃吀 device> [GvDevice::create_stream] Number of stream channels = 1
[18:28:03.052] 馃吀 device> [GvDevice::auto_packet_size] Current packet size check successfull (8164 bytes)
[18:28:03.054] 馃吀 stream> [GvStream::stream_new] Stream channel = 0
[18:28:03.056] 馃吀 stream> [GvStream::stream_new] Packet size = 8164 byte(s)
[18:28:03.059] 馃吀 stream> [GvStream::stream_new] Destination stream port = 50350
[18:28:03.059] 馃吀 stream> [GvStream::stream_new] Source stream port = 20202
[18:28:03.060] 馃吀 stream> [GvStream::loop] Standard socket method
gv packet size         = 8164 bytes
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s

EmmanuelP, 2025-01-23:
You have an issue with the software trigger, so you have to enable the software trigger input and generate a software trigger. This is configured using --trigger and --software-trigger options.

liyu9202, 2025-02-12:
where can i set the options? by arv-camera-test-0.8 command or by my camera settings? does it any errors with my attached code?