AravisProject/aravis #982: Can't compile 0.8.33 on Ubuntu 18.04 due to missing include

avsegal, 2025-02-12:
**Describe the bug**
Can't compile 0.8.33 on Ubuntu 18.04 due to missing include for errno variable.

**To Reproduce**
Steps to reproduce the behavior:
Compile aravis 0.8.33 on Ubuntu 18.04.6 LTS

**Platform description:**
 - Aravis version 0.8.33
 - OS: Ubuntu 18.04
 - Hardware x86_64

**Additional context**
Was able to fix by adding #include <errno.h> to the top of arvnetwork.c

henriquesimoes, 2025-02-14:
AFAIU, this is a duplicate of #948, which has already been fixed by ff96d8cacdbc1d0fd123409adb53397903f6e77d. As of now, there is no stable release containing the fix, though.

EmmanuelP, 2025-02-16:
Hi,

I have just released aravis 0.8.34 which include the fix.

closed 2025-02-16 by EmmanuelP