AravisProject/aravis #983: Error when running daheng gigevision camera

bug-programmer, 2025-02-14:
**Describe the bug**
A clear and concise description of what the bug is.

I have a Daheng Imaging-MER2-507-23GC-P GigE vision camera. It works on the arv-viewer but when using a simple gstreamer pipeline i get an error. 

**To Reproduce**
Gstreamer pipeline - 
`ARV_DEBUG=all gst-launch-1.0 aravissrc ! videoconvert ! ximagesinkSetting` 

```
[10:42:19.369] 🅸 interface> Found 0 USB3Vision device (among 19 USB devices)
[10:42:19.370] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[10:42:19.370] 🅸 interface> [GvDiscoverSocket::new] Add interface 172.18.0.1 (172.18.255.255)
[10:42:19.370] 🅸 interface> [GvDiscoverSocket::new] Add interface 172.17.0.1 (172.17.255.255)
[10:42:19.370] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.1.192 (192.168.1.255)
[10:42:19.370] 🅸 interface> [GvInterface::discovery] Device 'Daheng Imaging-MER2-507-23GC-P-' found (interface 192.168.1.192) user_id 'mycam' - MAC '00:21:49:f6:b6:8e'
[10:42:19.370] 🅸 device> [GvDevice::new] Interface address = 192.168.1.192
[10:42:19.370] 🅸 device> [GvDevice::new] Device address = 192.168.1.188
[10:42:19.373] 🅸 device> [GvDevice::load_genicam] xml url = 'Local:MER2_GIGE_V2.4.12.zip;41a30000;890a' at 0x200
[10:42:19.373] 🅸 device> [GvDevice::load_genicam] Xml address = 0x41a30000 - size = 0x890a - MER2_GIGE_V2.4.12.zip
[10:42:19.536] 🅸 device> [GvDevice::load_genicam] Zipped xml data
[10:42:19.537] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.538] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.539] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.539] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.540] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.540] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.540] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.542] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.542] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.542] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.543] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.543] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.543] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.544] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.544] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.544] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.544] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.545] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.546] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.546] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.546] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.546] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.546] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.546] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.546] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.546] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.546] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:42:19.547] 🅸 dom> [Genicam::create_element] Unknown tag (SwapEndianess)
[10:42:19.547] 🅸 genicam> [Gc::set_default_node_data] Add 'DeviceManufacturerInfo'
[10:42:19.547] 🅸 genicam> [Gc::set_default_node_data] Add 'DeviceID'
[10:42:19.547] 🅸 genicam> [Gc::set_default_node_data] Add 'GevTimestampTickFrequency'
[10:42:19.547] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPHostPort'
[10:42:19.547] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPSDoNotFragment'
[10:42:19.547] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPSBigEndian'
[10:42:19.547] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCSP'
[10:42:19.547] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPAddrCalc'
[10:42:19.548] 🅸 device> [GvDevice::new] Device endianness = big
[10:42:19.548] 🅸 device> [GvDevice::new] Packet resend     = yes
[10:42:19.548] 🅸 device> [GvDevice::new] Write memory      = yes
[10:42:19.548] 🅸 device> [GvDevice::new] Legacy endianness handling = yes

** (gst-launch-1.0:441855): CRITICAL **: 10:42:19.550: arv_camera_uv_set_usb_mode: assertion 'arv_camera_is_uv_device (camera)' failed
Setting pipeline to PAUSED ...
[10:42:19.578] 🆆 misc> [PixelFormat::to_gst_caps_string] 0x01100010 not found
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
0:00:00.219013586 441855 0x613b6501e060 WARN                 basesrc gstbasesrc.c:3127:gst_base_src_loop:<aravis0> error: Internal data stream error.
0:00:00.219028053 441855 0x613b6501e060 WARN                 basesrc gstbasesrc.c:3127:gst_base_src_loop:<aravis0> error: streaming stopped, reason not-negotiated (-4)
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Internal data stream error.
Additional debug info:
../libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstAravis:aravis0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.000168625
Setting pipeline to NULL ...
Freeing pipeline ...
```

**Expected behavior**
I figured it should work normally with GigE vision cameras too. Perhaps something wrong with the expected pixel format ? - i assume daheng is expecting something non-standard. 

**Camera description:**
 - Daheng Imaging
 - MER2-507-23GC-P
 - GigE vision
 - 
**Platform description:**
 - Aravis version: 0.8.20
 - OS: Ubuntu 22.04
 - Hardware: x86_64

**Additional context**
Works with the vendor's viewer - `Galaxy viewer` and `arv-viewer-0.8`

EmmanuelP, 2025-02-16:
What pixel formats are supported by your camera ?

bug-programmer, 2025-02-16:
Hello, Thank you for the response. 

The output of `arv-tool-0.8 features PixelFormat` is

```
Daheng Imaging-MER2-507-23GC (192.168.1.188)
        Enumeration : 'PixelFormat'
            EnumEntry   : 'BayerGR12'
            EnumEntry   : 'BayerGR8'
```

SzGaa, 2025-07-22:
Probably same as [Internal data stream error (Aravis + GStreamer + Daheng USB3) #866
](https://github.com/AravisProject/aravis/issues/866)

Use:
video/x-bayer, **format=rggb**