AravisProject/aravis #989: build fail on ubuntu 22.04

Ubiquite, 2025-02-18:
**Describe the bug**
the ninja install step is failing on a clear ubuntu 22.04

**To Reproduce**
build aravis on ubuntu 22.04:

```
autera-admin@autera-autobox:~/aravis-0.8.34/build$ ninja install
[0/1] Installing files
Installing subdir /home/autera-admin/aravis-0.8.34/viewer/icons/gnome to /usr/local/share/icons/hicolor
Installation failed due to insufficient permissions.
Attempt to use /usr/bin/sudo to gain elevated privileges? [y/n] y
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/mesonbuild/mesonmain.py", line 146, in run
    return options.run_func(options)
  File "/usr/lib/python3/dist-packages/mesonbuild/minstall.py", line 811, in run
    installer.do_install(datafilename)
  File "/usr/lib/python3/dist-packages/mesonbuild/minstall.py", line 564, in do_install
    d = self.check_installdata(pickle.load(ifile))
ModuleNotFoundError: No module named 'mesonbuild.options'

ERROR: Unhandled python exception

    This is a Meson bug and should be reported!
FAILED: meson-internal__install 
/home/autera-admin/.local/bin/meson install --no-rebuild
ninja: build stopped: subcommand failed.

```

**Platform description:**
 - Aravis version latest
 - OS: ubuntu 22.04
 - x86_64

EmmanuelP, 2025-02-18:
Hi,

Probably not an aravis issue: https://askubuntu.com/questions/1225195/meson-build-error-modulenotfounderror-no-module-named-mesonbuild

Ubiquite, 2025-02-18:
Ok thank you, I guess the teledyne documentation is out of date
https://www.teledynevisionsolutions.com/support/support-center/application-note/iis/using-the-aravis-gstreamer-source-plug-in/