AravisProject/aravis #999: Can't connect to 2 USB cameras at the same time, each one individually works fine

abrock, 2025-03-20:
**Describe the bug**

I have two USB-cameras (TIS DMK 38UX541) which used to work nicely with Aravis a couple of years ago.
Since then I re-installed my operating system (Debian stable) and now I can't use them anymore at the same time.
Each camera works nicely if I only one of them is attached to the computer, or if the program only tries to connect to one of them, regardless of which one and regardless of the choice of USB port.

When I call arv_camera_new on the second camera the GError* message is

"Error: Failed to claim USB control interface to '(null)-(null)-(null)-DMK 38UX541': LIBUSB_ERROR_BUSY"

It doesn't matter which of the two cameras I try to connect to first, it always fails for the second camera.

**To Reproduce**

1. Connect 2 USB-cameras to the computer
2. Compile and run [multi-device-test.txt](https://github.com/user-attachments/files/19363022/multi-device-test.txt) using a C++ compiler

**Expected behavior**

The program outputs (among other things)

Image successfully acquired for cam #0
Image successfully acquired for cam #1

**Camera description:**
 - Manufacturer: The Imaging Source Europe GmbH
 - Model: DMK 38UX541
 - Interface: USB3

**Platform description:**
 - Aravis version: e7e3560095ba5dc6834892f997fbeb3ef4d4e327 and 0.8.34
 - OS: Debian 12.10
 - Hardware: x86_64
 
/etc/udev/rules.d/aravis.rules 
```
# Give USB3Vision device access to everyone, and create a symlink in /dev/u3v
SUBSYSTEM=="usb", ATTRS{bDeviceClass}=="ef", ATTRS{bDeviceSubClass}=="02", ATTRS{bDeviceProtocol}=="01",
        ENV{ID_USB_INTERFACES}=="*:ef0500:*", MODE="0666", SYMLINK+="u3v/by-id/$env{ID_SERIAL}"

```

**Additional context**

I ran the following tests and attached log files for each:

[main-branch-1st-device.txt](https://github.com/user-attachments/files/19362970/main-branch-1st-device.txt) Main branch of aravis, connect to the first device only (OK)
[main-branch-2nd-device.txt](https://github.com/user-attachments/files/19362971/main-branch-2nd-device.txt) connect to the second device only (OK)
[main-branch-2devices.txt](https://github.com/user-attachments/files/19362967/main-branch-2devices.txt) connect to both devices (FAILURE)

[stable-1st-device.txt](https://github.com/user-attachments/files/19362972/stable-1st-device.txt) Release 0.8.34, connect to the first device only (OK)
[stable-2nd-device.txt](https://github.com/user-attachments/files/19362969/stable-2nd-device.txt) connect to the second device only (OK)
[stable-2devices.txt](https://github.com/user-attachments/files/19362968/stable-2devices.txt) connect to both devices (FAILURE)
[stable-inverse-order.txt](https://github.com/user-attachments/files/19362973/stable-inverse-order.txt) connect to both devices in reverse order (FAILURE)

I'm also using the same computer for controlling 17 GigE cameras (FLIR BFS) at the same time without issues.

EmmanuelP, 2025-04-21:
Hi,

Thanks for the report. I have tested your sample code with several USB3Vision devices. It works fine, unless a TheImagingSource camera is involved (a DFK 33UJ003).

When one of the 2 cameras is a TIS, sometimes it works reliably, sometimes not at all. When it does not work, I see the camera reset itself regularly in the kernel logs, even when the camera is not used by any software. It looks like a firmware issue. I have tried to disable the uvcvideo kernel module (these cameras also expose a v4l2 interface), in case there was a bad interaction, but it did not help.

Do you have camera from other manufacturer, in order to confirm it is a TIS only issue ?

EmmanuelP, 2025-04-21:
I have pushed to aravis main a slightly modified version of your sample code. The TIS camera is definitely the problematic one in my USB3Vision device collection:

```
pacaud in 🌐 lpm in Sources/aravis/build on  main [!] ❯ ./tests/arv-multi-uv-test
Aravis 0.9.0
id of #0: The Imaging Source Europe GmbH-199E41610435-41610435
vendor of #0: The Imaging Source Europe GmbH
model of #0: DFK 33UJ003
Image successfully acquired for camera #0
```

```
pacaud in 🌐 lpm in Sources/aravis/build on  main [!] took 2s ❯ ./tests/arv-multi-uv-test
Aravis 0.9.0
id of #0: Basler-2676014A8F61-21663585
vendor of #0: Basler
model of #0: acA1920-25um
id of #1: Basler-2676014EA111-21930257
vendor of #1: Basler
model of #1: acA1920-155um
id of #2: Basler-267601CA50CB-40209402
vendor of #2: Basler
model of #2: daA1440-220uc
id of #3: Basler-267601CB75AB-40257834
vendor of #3: Basler
model of #3: daA1920-160um
id of #4: FLIR-1E1000000001-19366937
vendor of #4: FLIR
model of #4: U3V camera
id of #5: Hikrobot-2BDF48899470-J48899470
vendor of #5: Hikrobot
model of #5: MV-CB013-A0UM-S
id of #6: Hikrobot-2BDF91459291-02G91459291
vendor of #6: Hikrobot
model of #6: MV-CB060-10UC-S
id of #7: Point Grey Research-1E1000F8916C-00F8916C
vendor of #7: Point Grey Research
model of #7: Blackfly S BFS-U3-13Y3C
id of #8: The Imaging Source Europe GmbH-DFK 33UJ003-41610435
vendor of #8: The Imaging Source Europe GmbH
model of #8: DFK 33UJ003
Test failed: USB device '(null):(null):(null)' not found
```

abrock, 2025-04-21:
I have two random USB cameras at home and verified that they work nicely on two different computers using the test program from the main branch:

```
Aravis 0.9.0
id of #0: Basler-267601CB8FC8-40536623
vendor of #0: Basler
model of #0: a2A2840-48umBAS
id of #1: Daheng Imaging-2BA200003795-FCU25010001
vendor of #1: Daheng Imaging
model of #1: MER2-1220-32U3C
Image successfully acquired for camera #0
Image successfully acquired for camera #1
```

I'll try to look into the firmware version, kernel logs, etc. of the TIS cameras this or next week.

abrock, 2025-04-21:
Small comment regarding the test program: Usage of tabs vs. spaces for indentation is inconsistent.

abrock, 2025-04-28:
> When it does not work, I see the camera reset itself regularly in the kernel logs, even when the camera is not used by any software.

Could you post the relevant parts of the kernel logs and how you get them? I don't see messages indicating camera reset using dmesg.

abrock, 2025-04-30:
I did some more tests with camera combinations:

1. I plug in one camera, kernel log:
```
[15989.362306] usb 1-1: new high-speed USB device number 8 using xhci_hcd
[15989.635197] usb 1-1: New USB device found, idVendor=199e, idProduct=981c, bcdDevice= 0.00
[15989.635203] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[15989.635206] usb 1-1: Product: DMK 38UX541
[15989.635208] usb 1-1: Manufacturer: The Imaging Source Europe GmbH
[15989.635210] usb 1-1: SerialNumber: 17220363
```
2. I run arv-multi-uv-test without problems with the single camera.
3. I plug in the second camera, kernel log:
```
[16021.713455] usb 1-2: new high-speed USB device number 9 using xhci_hcd
[16021.986921] usb 1-2: New USB device found, idVendor=199e, idProduct=981c, bcdDevice= 0.00
[16021.986927] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[16021.986930] usb 1-2: Product: DMK 38UX541
[16021.986932] usb 1-2: Manufacturer: The Imaging Source Europe GmbH
[16021.986933] usb 1-2: SerialNumber: 17220365
```
4. I run arv-multi-uv-test and it fails. No new kernel messages.

I also tried the following combinations which all work without problems:
- 1 TIS + 1 Daheng
- 1 TIS + 1 Basler
- 1 TIS + 1 Daheng + 1 Basler

The behaviour is the same regardless of which of my 3 TIS cameras I use.

EmmanuelP, 2025-04-30:
I'm not able to reproduce the issue currently. I just have one TIS camera in my collection.

abrock, 2025-04-30:
I found something interesting:
When connect 2 cameras and open arv-viewer I see two cameras listed. I can click on any one of them and click "Start video acquisition" and I will get images, but the images always come from the first camera (verified by putting a lens cap on the second camera). The title text of the image window shows different camera IDs depending on which camera I selected.

Recall that this is the error message when trying to connect to 2 cameras at the same time:
Error: Failed to claim USB control interface to '(null)-(null)-(null)-DMK 38UX541': LIBUSB_ERROR_BUSY

I'm speculating that the problem is something like this:
Aravis doesn't correctly pass each camera's unique identifier to libusb (therefore we see (null)-(null)-(null) in the message). libusb then just uses the first device again which is already in use.

I looked into the code a little bit and added some debug output code. This is my log now:

```
Aravis 0.9.0
id of #0: The Imaging Source Europe GmbH-DMK 38UX541-17220363
vendor of #0: The Imaging Source Europe GmbH
model of #0: DMK 38UX541
[20:55:04.839] 🅸 device> [UvDevice::new] Vendor  = NULL
[20:55:04.839] 🅸 device> [UvDevice::new] Product = NULL
[20:55:04.839] 🅸 device> [UvDevice::new] S/N     = NULL
[20:55:04.839] 🅸 device> [UvDevice::new] GUID    = DMK 38UX541
[20:55:04.935] 🅸 device> [UvDevice::new] Using control endpoint 2, interface 0
[20:55:04.935] 🅸 device> [UvDevice::new] Using data endpoint 3, interface 2
[20:55:04.935] 🅸 device> Get genicam
[20:55:04.936] 🅸 device> MANUFACTURER_NAME =        'The Imaging Source Europe GmbH'
[20:55:04.936] 🅸 device> MAX_DEVICE_RESPONSE_TIME = 0x0000012c
[20:55:04.936] 🅸 device> DEVICE_CAPABILITY        = 0x0000000000000e09
[20:55:04.936] 🅸 device> SRBM_ADDRESS =             0x4d52425300000000
[20:55:04.936] 🅸 device> MANIFEST_TABLE_ADDRESS =   0x5453464d00000000
[20:55:04.936] 🅸 device> U3VCP_CAPABILITY =         0x0000000000000001
[20:55:04.936] 🅸 device> MAX_CMD_TRANSFER =         0x00000400
[20:55:04.936] 🅸 device> MAX_ACK_TRANSFER =         0x00000400
[20:55:04.936] 🅸 device> SIRM_OFFSET =              0x4d52495300000000
[20:55:04.939] 🅸 device> SIRM_INFO =                0x04000000
[20:55:04.939] 🅸 device> SIRM_CONTROL =             0x00000000
[20:55:04.939] 🅸 device> SIRM_REQ_PAYLOAD_SIZE =    0x0000000001358a40
[20:55:04.939] 🅸 device> SIRM_REQ_LEADER_SIZE =     0x00000034
[20:55:04.939] 🅸 device> SIRM_REQ_TRAILER_SIZE =    0x00000020
[20:55:04.939] 🅸 device> SIRM_MAX_LEADER_SIZE =     0x00000040
[20:55:04.939] 🅸 device> SIRM_PAYLOAD_SIZE =        0x00100000
[20:55:04.939] 🅸 device> SIRM_PAYLOAD_COUNT =       0x00000013
[20:55:04.939] 🅸 device> SIRM_TRANSFER1_SIZE =      0x00058a40
[20:55:04.939] 🅸 device> SIRM_TRANSFER2_SIZE =      0x00000000
[20:55:04.939] 🅸 device> SIRM_MAX_TRAILER_SIZE =    0x00000020
[20:55:04.939] 🅸 device> MANIFEST_N_ENTRIES =       0x0000000000000001
[20:55:04.939] 🅸 device> MANIFEST ENTRY
                         00000000 62 05 00 01 00 04 01 01 00 00 00 c0 00 00 00 00  b...............
                         00000010 c8 3b 00 00 00 00 00 00 4e 1e f7 86 bf 89 98 61  .;......N......a
                         00000020 8a b8 b4 38 2b fe 1b 1c 17 c0 67 38 00 00 00 00  ...8+.....g8....
                         00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[20:55:04.939] 🅸 device> genicam address =          0x00000000c0000000
[20:55:04.939] 🅸 device> genicam size    =          0x0000000000003bc8
[20:55:05.162] 🅸 device> zip file =                 IMX541_M.xml
[20:55:05.371] 🅸 device> [UvDevice::new] Vendor  = NULL
[20:55:05.371] 🅸 device> [UvDevice::new] Product = NULL
[20:55:05.371] 🅸 device> [UvDevice::new] S/N     = NULL
[20:55:05.371] 🅸 device> [UvDevice::new] GUID    = DMK 38UX541
[20:55:05.466] 🅸 device> [UvDevice::new] Using control endpoint 2, interface 0
[20:55:05.466] 🅸 device> [UvDevice::new] Using data endpoint 3, interface 2
[20:55:05.466] 🅸 device> Get genicam
[20:55:05.467] 🅸 device> MANUFACTURER_NAME =        'The Imaging Source Europe GmbH'
[20:55:05.467] 🅸 device> MAX_DEVICE_RESPONSE_TIME = 0x0000012c
[20:55:05.467] 🅸 device> DEVICE_CAPABILITY        = 0x0000000000000e09
[20:55:05.467] 🅸 device> SRBM_ADDRESS =             0x4d52425300000000
[20:55:05.467] 🅸 device> MANIFEST_TABLE_ADDRESS =   0x5453464d00000000
[20:55:05.467] 🅸 device> U3VCP_CAPABILITY =         0x0000000000000001
[20:55:05.467] 🅸 device> MAX_CMD_TRANSFER =         0x00000400
[20:55:05.467] 🅸 device> MAX_ACK_TRANSFER =         0x00000400
[20:55:05.467] 🅸 device> SIRM_OFFSET =              0x4d52495300000000
[20:55:05.470] 🅸 device> SIRM_INFO =                0x04000000
[20:55:05.470] 🅸 device> SIRM_CONTROL =             0x00000000
[20:55:05.470] 🅸 device> SIRM_REQ_PAYLOAD_SIZE =    0x0000000001358a40
[20:55:05.470] 🅸 device> SIRM_REQ_LEADER_SIZE =     0x00000034
[20:55:05.470] 🅸 device> SIRM_REQ_TRAILER_SIZE =    0x00000020
[20:55:05.470] 🅸 device> SIRM_MAX_LEADER_SIZE =     0x00000040
[20:55:05.470] 🅸 device> SIRM_PAYLOAD_SIZE =        0x00100000
[20:55:05.470] 🅸 device> SIRM_PAYLOAD_COUNT =       0x00000013
[20:55:05.470] 🅸 device> SIRM_TRANSFER1_SIZE =      0x00058a40
[20:55:05.470] 🅸 device> SIRM_TRANSFER2_SIZE =      0x00000000
[20:55:05.470] 🅸 device> SIRM_MAX_TRAILER_SIZE =    0x00000020
[20:55:05.470] 🅸 device> MANIFEST_N_ENTRIES =       0x0000000000000001
[20:55:05.470] 🅸 device> MANIFEST ENTRY
[... cut ...]
                         00000010 c8 3b 00 00 00 00 00 00 4e 1e f7 86 bf 89 98 61  .;......N......a
                         00000020 8a b8 b4 38 2b fe 1b 1c 17 c0 67 38 00 00 00 00  ...8+.....g8....
                         00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[20:55:05.470] 🅸 device> genicam address =          0x00000000c0000000
[20:55:05.470] 🅸 device> genicam size    =          0x0000000000003bc8
[20:55:05.692] 🅸 device> zip file =                 IMX541_M.xml
Test failed: Failed to arv_uv_device_get_instance_private()
```

abrock, 2025-04-30:
New theory:

```
GUID    = DMK 38UX541
```
The problem is that this is the same GUID for all my TIS cameras.
For my other cameras the GUID is much more "serial-number-like" looking, e.g. my Basler has `767601CB8FC8` and my Daheng `2BA200003795`, both of which are not similar to their respective model names `a2A2840-48umBAS` and `MER2-1220-32U3C`.

EmmanuelP, 2025-05-02:
You've found the issue. Let's forget about the kernel logs I have seen, it was probably an unrelated thing.

GUID stands for Globally unique identifier. Aravis is relying on it to find the device in the USB device tree. It used to rely on manufacturer + product + serial, that explains why it worked before. That was changed in commit b9c585f966e2325a6544762352be8e8b48ef3dc6 .

It is a firmware issue that should be reported to TIS.

A possible workaround would be to look for duplicated GUIDs, and fallback to 'manufacturer + product + serial' for these devices.

abrock, 2025-05-02:
I checked out the parent commit of https://github.com/AravisProject/aravis/commit/b9c585f966e2325a6544762352be8e8b48ef3dc6 and added a minor fix to the meson build file:
https://github.com/abrock/aravis/tree/tis-fix

Using this version of Aravis I'm now able to get images from 2 TIS cameras without any problems.
I'll contact the manufacturer and point them to this discussion.

abrock, 2025-05-07:
I received an experimental firmware from the manufacturer and installed it on my cameras. Now the arv-multi-uv-test succeeds. I found a different problem (not a bug in Aravis):

The problem with the wrong GUID only appeared when using a type of USB-cable which caused the camera to use USB2 instead of USB3. I wasn't aware of that problem and it lead to me drawing wrong conclusions (I thought one of the cameras was behaving better, but it just had a better cable). I'd like to put checks into my recording software which tell me if the camera uses a USB mode other than USB3, does Aravis provide functions to get the USB mode?

EmmanuelP, 2025-05-07:
Aravis does not ensure a device is connected using USB3, as I think it should also work as USB2, just with a lower bandwidth.

I have pushed a patch to main that reads the device bcdUSB descriptor field and emit a warning debug if the version is lower than 3.