AravisProject/aravis #1005: write-protect error when trying to set the ImagingMode parameter

sergioenriquez, 2025-04-18:
**Describe the bug**
Using arv-tool to set the value of the ImagingMode parameter gives a "GigEVision write_memory error (write-protect)" error message

**To Reproduce**
Call this from the command line:
arv-tool-0.8 control ImagingMode=Capture3D

**Expected behavior**
Imaging mode setting change is accepted and set to Capture3D

**Camera description:**
 - Manufacturer: Keyence
 - Model: VJ-3002 Image processing unit with a CA-H500CX camera 
 - Interface: GigEVision 

**Platform description:**
 - Aravis version: 0.8
 - OS: Ubuntu 22.04
 - Hardware:  x86_64

**Additional context**
Output from the command line:
$ arv-tool-0.8 control ImagingMode=Capture3D --debug=interface:3,device:3,stream:3,stream-thread:3
[15:25:24.015] 🅸 interface> Found 0 USB3Vision device (among 4 USB devices)
[15:25:24.015] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[15:25:24.016] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.50.79 (192.168.51.255)
[15:25:24.016] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.0.10 (192.168.0.255)
[15:25:24.051] 🅸 interface> [GvInterface::discovery] Device 'KEYENCE CORPORATION-CA-H500CX-AX420071' found (interface 192.168.0.10) user_id 'SHD' - MAC '00:01:fc:f0:28:1e'
KEYENCE CORPORATION-CA-H500CX-AX420071 (192.168.0.100)
[15:25:25.052] 🅸 interface> Found 0 USB3Vision device (among 4 USB devices)
[15:25:25.052] 🅸 device> [GvDevice::new] Interface address = 192.168.0.10
[15:25:25.052] 🅸 device> [GvDevice::new] Device address = 192.168.0.100
[15:25:25.053] 🅸 device> [GvDevice::load_genicam] xml url = 'local:Keyence_HX-xxx_v2_0_8.zip;38000000;21DF7' at 0x200
[15:25:25.053] 🅸 device> [GvDevice::load_genicam] Xml address = 0x38000000 - size = 0x21df7 - Keyence_HX-xxx_v2_0_8.zip
[15:25:25.208] 🅸 device> [GvDevice::load_genicam] Zipped xml data
[15:25:25.263] 🅸 device> [GvDevice::new] Device endianness = big
[15:25:25.263] 🅸 device> [GvDevice::new] Packet resend     = yes
[15:25:25.263] 🅸 device> [GvDevice::new] Write memory      = yes
[15:25:25.263] 🅸 device> [GvDevice::new] Legacy endianness handling = no
ImagingMode write error: GigEVision write_memory error (write-protect)
(pyenv) nuvo-8108gc@BLA05-SP4-Nuvo-8108GC:~$

EmmanuelP, 2025-04-20:
Maybe the write access is conditioned by the value of another feature. What does the camera manual say ?

sergioenriquez, 2025-04-21:
The manual says that the device needs to be in setup mode before trying to change ImagingMode, so I tried doing this by calling `arv-tool-0.8 control OperationMode=SetupMode` and this setting change seemed to be accepted, but it turns out that the camera was reverting to RunMode automatically when a new connection was made to it, so that is why it did not let me to set ImagingMode. 

To be able to change the imaging mode with the command line tool we needed to combine these parameters changes like this:
`arv-tool-0.8 control OperationMode=SetupMode ImagingMode=Capture3D`

EmmanuelP, 2025-04-21:
Does that mean you solved the issue and this ticket can be closed ?

sergioenriquez, 2025-04-21:
Yes, we are able to change the imaging mode now.

closed 2025-04-21 by EmmanuelP