AravisProject/aravis #1011: Using GStreamer aravissrc with LUCID Atlas ATP051S-CC and Hardware Trigger

JQuesenberry, 2025-05-19:
Hello,

I'm working with a LUCID Atlas ATP051S-CC, I'm able to capture video from it using the aravissrc element, however I am not having any luck trigger image capture using property trigger=Line2. The property is very open-ended and gives to valid entry options, though from looking through the various bits of codes it seems like Software, Line0-8, Action0, and Encoder0 might all be valid options. As mentioned, I've tried both trigger=Line2 and trigger=line2 and the images are still coming in at a rate much higher than the hardware trigger is running, which is 1 Hz since that's easily noticeable to be working or not.

The aravissrc element was built from the main branch and the latest commit was at that moment d3817cae09f75067901a6dde842a5a280bd95261 Mon Apr 21 09:56:20 2025 +0200.

Here's an example pipeline of what I'm trying:

gst-launch-1.0 -v -e \
        aravissrc camera-name=10.10.64.20 exposure-auto=off exposure=8000 gain-auto=off gain=0 h-binning=1 v-binning=1 trigger=Line2 ! \
        video/x-bayer,format=rggb,pixel-aspect-ratio=1/1,width=2448,height=2048,framerate=0/1 ! \
        bayer2rgb ! \
        videoconvert ! \
        videoscale ! \
        video/x-raw,width=1280 ! \
        ximagesink

Can someone walk me through what I might be missing?

Also, if I were to choose trigger=Software, is there anyway to issue that trigger when using gst-launch-1.0? Or would I need to put the whole pipeline into a Python script to gain access to that functionality? I will ultimately be the GStreamer Daemon, so that complicates things I imagine.

Thanks,

JQ

EmmanuelP, 2025-05-20:
In the documentation:

```
// Connect to camera
// Get device node map 

// Choose Line0 and set it to Input 
LineSelector = Line0;
LineMode = Input;

TriggerMode = On;
TriggerSelector = FrameStart;           // Trigger signal starts a frame capture 
TriggerSource = Line0;                  // External signal to be expected on Line0 
TriggerActivation = FallingEdge;        // Camera will trigger on the falling edge of the input signal 

AcquisitionStart();

// Acquire images by sending pulses to Line0
```

You may have to set the line to input, using `aravissrc` `features` parameter.

JQuesenberry, 2025-06-04:
Regarding Aravis unexpectedly overriding settings I've manually set using arv-tool-0.10 before running GStreamer ... I've added debug messages to gstaravis.c and I see the following when inspecting values...

During gst_aravis_init_camera, right after arv_camera_new:
GevSCPSPacketSize=9000
LineSelector=Line0
LineMode=Input
PtpEnable=true
PtpSlaveOnly=true
ExposureAuto=Off
TriggerSource=Line0
**TriggerSelector=FrameStart**
**TriggerActivation=RisingEdge**
**TriggerOverlap=PreviousFrame**
**TriggerMode=On**
AcquisitionFrameRate=60.507050
BinningHorizontalMode=Average
BinningVerticalMode=Average

On the first call to gst_aravis_create:
GevSCPSPacketSize=9000
LineSelector=Line0
LineMode=Input
PtpEnable=true
PtpSlaveOnly=true
ExposureAuto=Off
TriggerSource=Line0
**TriggerSelector=ExposureActive**
**TriggerActivation=LevelHigh**
**TriggerOverlap=Off**
**TriggerMode=Off**
AcquisitionFrameRate=60.002399
BinningHorizontalMode=Average
BinningVerticalMode=Average

It seems like this happens when I specify a framerate in the CAPS, whereas if I set the framerate to `0/1` then the settings are retained. I see spots in the Aravis code where it forces `TriggerMode=Off` and I'm guessing maybe this action causes the other Trigger values above to revert to defaults?

This doesn't explain the other problems I mentioned previously that I need help solving, but it seems like a step in the right direction.

JQuesenberry, 2025-06-06:
Is there anyone that can help me with this? I've spent a significant amount of time trying to get all the pieces to play well together and I'm blocked on progress at this point.