AravisProject/aravis #1012: Return error when control is not acquired

hgaiser, 2025-05-21:
It appears to me that when Aravis fails to acquire control of the camera, it will ignore this error and continue as if control was acquired. This is is visible in [this](https://github.com/AravisProject/aravis/blob/aba805c1c69c4e9d0351da556f08690c9a4f9960/src/arvgvdevice.c#L2056) line, where the `success` return value from `arv_gv_device_take_control` is ignored.

I ran into permission denied errors when setting the `TriggerMode` in my application, which appeared to be caused by never having acquired control over the camera (there was another application that had taken control). Instead of producing an error about acquiring control, it was producing permission denied errors about changing `TriggerMode`.

Is there a valid use case for ignoring errors when acquiring control? Or should it produce an error in that case?

EmmanuelP, 2025-05-21:
Aravis should not necessarily return an error if it fails to acquire control, as we may want to be able to get read acccess if another client has already taken control.

But I agree the implementation is not as clean as it should.

hgaiser, 2025-05-21:
Thanks for the quick reply. So the suggested approach is to check if control was acquired after setting up a camera and deal with it accordingly (assuming the application requires control)?

EmmanuelP, 2025-05-22:
`arv_device_[take|leave]_control()` ?